STMicroelectronics/STM32CubeF4 #1: Source code from STM32CubeF4 package does not exist

hooddanielc, 2019-05-11:
**Setup**
 * Latest version of git
 * Any compiler

**Describe the bug**
Unable to compile code

**How To Reproduce**
1. Clone the repository: `git clone https://github.com/STMicroelectronics/STM32CubeF4.git`
2. Locate any `.c` file

**Unexpected Behavior**
Unable to find any source code containing  the `STM32CubeF4` package.

**Expected Behavior**
Repository should contain latest version of code from `STM32CubeF4`. The source code should match download from [STMicroelectronics website tools and software part number STM32CubeF4](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-mcu-mpu-packages/stm32cubef4.html)

**Screenshots**
[image]

ALABSTM, 2019-05-13:
Dear Daniel Hood,

Thank you for your interest. We are presently working on this. We aim to have all STM32Cube packages available on Github soon.

We cannot share a date for the moment. We will keep you informed.

With regards

ALABSTM, 2019-08-05:
Dear Daniel,

STM32CubeF4 package (v1.24.1) has just been published on GitHub. We hope this addresses your request. In such case this issue could be closed. Thank you for your patience.

With regards,

hooddanielc, 2019-08-06:
lgtm

closed 2019-08-06 by hooddanielc