STMicroelectronics/STM32CubeF4 #2: SD_diskio_dma_template incorrect code

betgear, 2019-08-06:
Incorrect bracing in the sd_read and sd_write functions cause memory access alignment crashes when ENABLE_SCRATCH_BUFFER is enabled.

Additionally the memcpy is incorrect, copying data in the wrong direction int the write function.

I posted information about this in the forum linked

https://community.st.com/s/question/0D50X0000ArVPjuSQG/hal-bug-sddiskiodmatemplate-in-stm32cubefwf4v1241

As pull requests are not currently activated, I've fixed the issue and the code is attached in the .txt file.

[fixed.txt](https://github.com/STMicroelectronics/STM32CubeF4/files/3471427/fixed.txt)

[orig.txt](https://github.com/STMicroelectronics/STM32CubeF4/files/3471438/orig.txt)

I've attached the original code too, so a simple diff will reveal the code changes.

These problems also exist in the F7 library, I've not checked others.

ALABSTM, 2019-08-07:
Dear Betgear,

Thank you for pointing out this issue. It has been transferred to our development teams for deeper analysis. We will be back to you as soon as we get an answer. Thank you for you patience.

With regards,

betgear, 2019-08-14:
Hi Alab,

Any news on this?

CCASTM, 2019-08-21:
Hello Betgear,
Ali is out of office currently.
The development team is currently busy. I will remind them to pay attention to your Issue.
Regards
Christophe

ALABSTM, 2019-09-10:
Hi Betgear,

Our development teams confirmed both issues you pointed out:
1. The incorrect bracing.
2. The incorrect use of memcpy().
Both will be fixed and made available in a future release. The same thing will apply in case of other impacted products.

Thank you once more for your contribution and for your patience.

With regards,

ALABSTM, 2019-09-10:
ST Internal Reference: 71963

ALABSTM, 2019-09-10:
ST Internal Reference: 71973

ALABSTM, 2019-09-11:
Hi Betgear,

Pending the next STM32CubeF4 FW release, please find attached the [sd_diskio_dma_template.c.txt](https://github.com/STMicroelectronics/STM32CubeF4/files/3601617/sd_diskio_dma_template.c.txt)
 with the fixes that will be applied.

Remove the ".txt" file extension and replace your current sd_diskio_dma_template.c with this one. We hope this solves the problem you faced.

With regards,

betgear, 2019-09-12:
Hi Ali,

Thanks for that.

Should I close the issue, or leave it open until a new release of the FW so that other people can find the solution?

ALABSTM, 2020-02-28:
Hi Betgear,

I hope you are doing well. This issue has been fixed in the STM32CubeF4 **v1.25.0** firmware package which will be published soon (already available on our website "st.com").

With regards,

ronitsrma16, 2020-12-04:
Hi ALABSTM,

I'm using sd card with FATFS and freertos. My program is also getting crashed and I believe I can resolve it by following the above-mentioned procedure. 

I'm using **stm32cubeFW_F4 v1.24.1**, I downloaded the patch25.2 as well. I don't know how to install this patch in the current project and I didn't find any document also to do this. I tried to update the files in my project directly but inside the **patch folder**,  the middleware, project, utilities, and documentation folder is empty. 
So I unable to paste the file as well.

I tried placing the memcpy function and buff+=BLOCKSIZE line(2 lines), to the above  **BSP_SD_WriteBlocks_DMA**  function. But no success. Below is the snapshot of the sd card code with fatfs and freertos. 
Please suggest which lines I should place above the DMA function.

`
 for (i = 0; i < count; i++)
 {
      ret = BSP_SD_WriteBlocks_DMA((uint32_t*)scratch, (uint32_t)sector++, 1);
      if (ret == MSD_OK)
   {
        /* wait for a message from the queue or a timeout */
        event = osMessageGet(SDQueueID, SD_TIMEOUT);

        if (event.status == osEventMessage)
      {
          if (event.value.v == WRITE_CPLT_MSG)
         {
            memcpy((void *)buff, (void *)scratch, BLOCKSIZE);
            buff += BLOCKSIZE;
          }
        }
      }
      else
      {
        break;
      }
 }
`

CCASTM, 2020-12-04:
Hello ronitsrma16

In fact you don't need to download the patch v1.25.2 and merge it but rather to go back to GitHub , select the Tag 1.25.2 and launch a git pull 

Regards
Christophe

ronitsrma16, 2020-12-05:
Hi CCASTM,

I unable to do a git pull cube IDE. can you share any document about it or if you can tell, which lines I have to modify the code I've posted in order to make it work?

DeeFuse, 2022-04-26:
Is this issue still beeing worked on?
The current CubeMX code sill generates misplaced brackets when unsing ENABLE_SCRATCH_BUFFER, rendering the DMA diskio useless.
Without it, unaligned writes will write garbage to the disk when the src buffer is unaligned.

ALABSTM, 2022-04-29:
Hi @DeeFuse,

This issue has been closed since a while as **fixed** in the frame of version **1.25.0** of this firmware. In case you noticed something wrong, please report it in a new thread and we will have a look at it.

With regards,