STMicroelectronics/STM32CubeF4 #3: Bit position of TimeFormat in HAL_RTC

windingcat, 2019-09-06:
I am developing the RTC function with HAL library on STM32F4. There is a confuss on the bit position of timeformat for RTC register operation.

In the source code, the TimeFormat position of HAL_RTC_SetTime is **16**:

[image]

Yet in the Reference manual (RM0430, page 760), it says:
[image]

It happens on other functions, such as HAL_RTC_GetTime, HAL_RTC_GetAlarm, HAL_RTC_SetAlarm.
Could you share why the bit position of TimeFormat is 16?

ALABSTM, 2019-09-06:
Hi windingcat,

First, let me thank you for your contribution. Actually, both reference manual and RTC HAL driver remain coherent although it may not seem the case at first glance.

Indeed, parameter "TimeFormat" should be assigned one out of the two predefined values listed below:
https://github.com/STMicroelectronics/STM32CubeF4/blob/b5abca20c9676b04f8d2885a668a9b653ee65705/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_rtc.h#L269-L270
As you may notice, constant "RTC_HOURFORMAT12_PM" is defined to be 0x40 which corresponds to a 6-time shift (0x01 << 6).

Added to this the 16-time shift you highlighted, the total yields a 22-time shift, exactly as indicated in the reference manual.
https://github.com/STMicroelectronics/STM32CubeF4/blob/b5abca20c9676b04f8d2885a668a9b653ee65705/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc.c#L750
I agree this might be confusing a little bit but the driver handles this parameter correctly.

I hope this answers your question and thank you once more for pointing out this detail.

With regards,

windingcat, 2019-09-09:
hi, ALABSTM, Thanks for your kind response. 
It is good news that it is working well. Although from my side, I prefer to match the hardware manual instead of this trick since there may be extra on maintaining codes.

closed 2019-09-09 by windingcat