STMicroelectronics/STM32CubeF4 #4: UART Baudrate Calculation Error At Clock Higher Than 172MHz

resetcoder, 2019-10-08:
Problem: STM32F407 can not calculate correct baud rate if MCU is overclocked over 171MHz.

UART Baudrate calculation happens in (i.e.) `stm32f4xx_hal_uart.c` in function `UART_SetConfig(UART_HandleTypeDef *huart)`.

Inside this function the following line calculates the `DIV_Mantissa` and `DIV_Fraction` values for the UART `BRR` register, which sets up a clock divider for the serial port to be as close as possible to the desired baud rate clock:

`huart->Instance->BRR = UART_BRR_SAMPLING8(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);`

(There is 16 bit caltulation also, but it is irrelevant for now.)

The function that returns the register value as result is a macro (in `stm32f4xx_hal_uart.h`).
Inside this macro the return value of `HAL_RCC_GetPCLK2Freq()` is multiplied by (uint)25, (uint)100, divided by (uint)100, etc.

The return value of `HAL_RCC_GetPCLK2Freq()` is the system clock frequency in Hz. In a 168MHz environment this is a relatively huge number for the (uint32_t) return type, because this number is later multiplied, divided by the (uint) numbers inside the macro.

My observation is if the system clock frequency is the following:

`UART_BRR_SAMPLING8((uint32_t)171798691, (uint32_t)115200)` returns `2979`,
`UART_BRR_SAMPLING8((uint32_t)171798692, (uint32_t)115200)` returns `0`

This is clearly an overflow, and system clock above 171.8MHz can not calculate the uart baudrate divider.

`UART_BRR_SAMPLING8((uint64_t)171798692, (uint32_t)115200)` returns `2979` again.

Changing the type from 32 bit uint to 64 bit uint of the first parameter fixes this issue, but getting the clock is coded inside the library and `HAL_RCC_GetPCLK2Freq()` return uint32_t.

This might be fixed systemwide by changing the return type of `HAL_RCC_GetPCLK2Freq()` to `uint64_t` or by changing the macro to force type casting for the first input parameter to 64 bit uint (but then this will fix only the uart related problems).

For now I fix this issue by changing the return type of `HAL_RCC_GetPCLK2Freq()` to 64 bit, but later I'd like to use the library updates, so please fix this issue if possible or tell me the reason why this function returns such type that may overflow, and what other method should I use then.

Thank You!

ALABSTM, 2019-10-10:
Hi resetcoder,

We transferred this issue to our development teams. We will be back to you as soon as we get an answer. Thank you for your contribution.

With regards,

GCASTM, 2019-10-11:
Hi @resetcoder 
I confirm issue you raise is present in STM32F4 FW package. We will track issue internally and ensure that it will be addressed in next F4 package delivery.
I will keep you posted here about implemented solution.
Regarding root cause, it appears that using uint32_t in this macro, was ok ok till F4 devices were operating at 168 Mhz max. Now it is cleared it has to be revised ...
Thanks for your valuable inputs.
Regards
GCA

resetcoder, 2019-10-11:
Hi @GCASTM !
Thank You for the answer. I know this particular device is not rated to be driven over 168MHz (even though it can be), but I'm not sure what other devices in the F4 series can run over 172MHz, this is why I issued this problem.
I'm working with STM devices only for a week now and this took me some time to figure out why I can not debug via uart when I slightly overclock it.
This week my F767 will arrive and continue some projects on that device and I've just noticed the same function returns 32 bit uint in the F7 HAL driver too, even though the base clock frequency is 216MHz (don't remember exactly). Please check also that F7 package for the same problem, because this may be an issue, but I could not confirm yet, but it worth a checking.
Greets

ALABSTM, 2019-10-16:
ST Internal Reference: 73894

GCASTM, 2019-10-21:
Hi @resetcoder 
On STM32F4xx serie, BRR baud rate calculation is specific to U(S)ART IP. Then, problem you raised, will be corrected in all HAL and LL drivers, that are using this peripheral (HAL_UART, HAL_USART, HAL_IRDA and HAL_SMARTCARD + LL_USART).

For HAL UART BRR case, here is the correction i would propose (in stm32f4xx_hal_uart.h) : 
`
#define UART_DIV_SAMPLING16(_PCLK_, _BAUD_)            ((uint32_t)((((uint64_t)(_PCLK_))*25U)/(4U*((uint64_t)(_BAUD_)))))
#define UART_DIVMANT_SAMPLING16(_PCLK_, _BAUD_)        (UART_DIV_SAMPLING16((_PCLK_), (_BAUD_))/100U)
#define UART_DIVFRAQ_SAMPLING16(_PCLK_, _BAUD_)        ((((UART_DIV_SAMPLING16((_PCLK_), (_BAUD_)) - (UART_DIVMANT_SAMPLING16((_PCLK_), (_BAUD_)) * 100U)) * 16U) + 50U) / 100U)
/* UART BRR = mantissa + overflow + fraction
            = (UART DIVMANT << 4) + (UART DIVFRAQ & 0xF0) + (UART DIVFRAQ & 0x0FU) */
#define UART_BRR_SAMPLING16(_PCLK_, _BAUD_)            ((UART_DIVMANT_SAMPLING16((_PCLK_), (_BAUD_)) << 4U) + \
                                                        (UART_DIVFRAQ_SAMPLING16((_PCLK_), (_BAUD_)) & 0xF0U) + \
                                                        (UART_DIVFRAQ_SAMPLING16((_PCLK_), (_BAUD_)) & 0x0FU))

#define UART_DIV_SAMPLING8(_PCLK_, _BAUD_)             ((uint32_t)((((uint64_t)(_PCLK_))*25U)/(2U*((uint64_t)(_BAUD_)))))
#define UART_DIVMANT_SAMPLING8(_PCLK_, _BAUD_)         (UART_DIV_SAMPLING8((_PCLK_), (_BAUD_))/100U)
#define UART_DIVFRAQ_SAMPLING8(_PCLK_, _BAUD_)         ((((UART_DIV_SAMPLING8((_PCLK_), (_BAUD_)) - (UART_DIVMANT_SAMPLING8((_PCLK_), (_BAUD_)) * 100U)) * 8U) + 50U) / 100U)
/* UART BRR = mantissa + overflow + fraction
            = (UART DIVMANT << 4) + ((UART DIVFRAQ & 0xF8) << 1) + (UART DIVFRAQ & 0x07U) */
#define UART_BRR_SAMPLING8(_PCLK_, _BAUD_)             ((UART_DIVMANT_SAMPLING8((_PCLK_), (_BAUD_)) << 4U) + \
                                                        ((UART_DIVFRAQ_SAMPLING8((_PCLK_), (_BAUD_)) & 0xF8U) << 1U) + \
                                                        (UART_DIVFRAQ_SAMPLING8((_PCLK_), (_BAUD_)) & 0x07U))
`

UART_DIV_SAMPLING16 and UART_DIV_SAMPLING8 contains the modification which address the problem. Added parenthesis in UART_BRR_SAMPLING16 and UART_BRR_SAMPLING8 aim to solve MISRA warnings.

To answer your last question on F7, BRR computation in more recent version of USART IP is slightly different, and macros are already able to handle higher frequencies. So no issue on F7/H7 in my opinion.

Any feedback welcome.
Regards
GCA

resetcoder, 2019-10-22:
Hello @GCASTM 
Thank You for working on this fix. The proposal seems fine to me, although I can not test it for now, but the forced type casting for the macro inputs to use them as 64 bit uint has fixed the problem for me, so I think this will fix it for sure, especially if You have tested it.

If the line `UART_BRR_SAMPLING8((uint32_t)171798692, (uint32_t)115200)` returns 2979 instead of 0, then it is probably working well.

I have no any experience with the F7 series and have no chance to test it, because unfortunately my F7 dev board got lost during the delivery process, I have to order another. But if You say the macros are different in that package then I believe it, because I haven't checked the macros, only the clock freq returning function.

As soon as possible when I get to my computer I'm working on I change the macros to what You have posted, and will check If everything works, but I don't think there will be any problem, since You have implemented the 64 bit casting for the input values.

Greets

ALABSTM, 2019-11-25:
Hi ResetCoder,

The fix you requested has been implemented and will be available in the frame of the next patch release, which will be published soon, hopefully. This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2019-11-25 by ALABSTM

ALABSTM, 2020-03-02:
Hi ResetCoder,

Just for the precision, the fix is available within the last STM32CubeF4 package release, i.e. **v1.24.2**.

With regards,