STMicroelectronics/STM32CubeF4 #5: Missed UARTs in BRR configuration if statement

CavKratos, 2019-10-21:
**Describe the set-up**
Board: 
     STM32F413 Nucleo-144
Software:
    STM32Cube_FW_F4_V1.24.1
    STMCubeMX version 5.4.0
    Atollic TrueSTUDIO® Version: 9.3.0 Build id: 20190212-0734

**Describe the bug**
In the if statement which sets configures the BRR (in stm32f4xx_hal_uart.c starting at line 3071) there are special cases for USART1 and USART 6 (because they're clocked from Peripheral Bus 2 (PB2)) but UART9 and UART10 (which are also clocked from PB2) are not included in this special case. This results in the baud rate of these UARTs being calculated from the clock rate of PB1 and so the baud rate of these 2 UARTS comes out as (desired baud rate)*(PB1 clock frequency)/(PB2 clock frequency)

**How To Reproduce**
Start a new project in STM32CubeMX loaded with STM32Cube_FW_F4_V1.24.1.
Initialise UART9 or UART10 in Asynchronous Mode.
Generate project.
Add "HAL_UART_Transmit(&huart9, (uint8_t*)"Hello\r\n", 7, HAL_MAX_DELAY);" or "HAL_UART_Transmit(&huart10, (uint8_t*)"Hello\r\n", 7, HAL_MAX_DELAY);" into the superloop in main(). Depending on which UART you have initialised.
Connect an STM32F413 Nucleo-144 and make a serial connection with pin PG1/ Nucleo header D64 (UART 9) or pin PE3/ Nucleo header PE3 (UART 10).
Open up a terminal at 115200 baud and connect to the board.
Debug the project.
This should print out gibberish.
If the baud is set to 230400 or the clock frequency of PB2 is set to 48MHz then it will repeatedly print out Hello as it should.

**Additional context**
I suggest that the instances of UART 9 and UART10 are added to the conditions of these 4 if statements.
Proposed new code (stm32f4xx_hal_uart.c starting at line 3071):
```
  /* Check the Over Sampling */
  if (huart->Init.OverSampling == UART_OVERSAMPLING_8)
  {
    /*-------------------------- USART BRR Configuration ---------------------*/
#if defined(USART6)
    if ((huart->Instance == USART1) || (huart->Instance == USART6) || (huart->Instance == UART9) || (huart->Instance == UART10))
    {
      huart->Instance->BRR = UART_BRR_SAMPLING8(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);
    }
#else
    if ((huart->Instance == USART1) || (huart->Instance == UART9) || (huart->Instance == UART10))
    {
      huart->Instance->BRR = UART_BRR_SAMPLING8(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);
    }
#endif /* USART6 */
    else
    {
      huart->Instance->BRR = UART_BRR_SAMPLING8(HAL_RCC_GetPCLK1Freq(), huart->Init.BaudRate);
    }
  }
  else
  {
    /*-------------------------- USART BRR Configuration ---------------------*/
#if defined(USART6)
    if ((huart->Instance == USART1) || (huart->Instance == USART6) || (huart->Instance == UART9) || (huart->Instance == UART10))
    {
      huart->Instance->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);
    }
#else
    if ((huart->Instance == USART1) || (huart->Instance == UART9) || (huart->Instance == UART10))
    {
      huart->Instance->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK2Freq(), huart->Init.BaudRate);
    }
#endif /* USART6 */
    else
    {
      huart->Instance->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), huart->Init.BaudRate);
    }
  }
```

**Screenshots**
Current state of the if statement:
[image]

Suggested change:
[image]

ALABSTM, 2019-10-29:
Hi CavKratos,

Thank you for this detailed issue. It will be forwarded to our development teams to evaluate its relevancy and, in case it is confirmed, the solution you suggested. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

Wit regards,

GCASTM, 2019-11-04:
Hi @CavKratos ,

Thanks for your inputs.
I confirm issue you noticed. For your information, a ticket was already internally raised for this problem, so we already have a reference for tracking the resolution till next F4 delivery.

Code modification you proposed is suitable in my opinion for your device.

Official F4 HAL code will be updated as follows in order : 
- to ensure solution is working for all F4 devices
- to also implement another change request about not calling UART_BRR_SAMPLING macros with a function as parameter (as this was leading to have the function executed several times).

Please find below a proposal of what could be the F4 HAL UART code for UART_SetConfig() function.
(in attached file)
[UART_SetConfig.txt](https://github.com/STMicroelectronics/STM32CubeF4/files/3804993/UART_SetConfig.txt)

Please let me know if this implementation also solves your problem.
Regards
GCA

ALABSTM, 2019-11-05:
ST Internal Reference: 50651

ALABSTM, 2019-11-25:
Hi CavKratos,

The fix you requested has been implemented and will be available in the frame of the next patch release, which will be published soon, hopefully. This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2019-11-25 by ALABSTM