STMicroelectronics/STM32CubeF4 #6: Misprint in USB Virtual-Port function brief

mlv-dev, 2019-11-19:
STM32CubeMX
(OS: Windows 10)

Target: F405
Enabled USB_OTG_FS & USB_DEVICE (VIRTUAL_PORT)
File: *project*\USB_DEVICE\App\usbd_cdc_if.c
Line: 255 (brief of CDC_Receive_FS fun.)
"untill exiting this function. If you exit this function before transfer"

double l in "utilll"

ALABSTM, 2019-11-21:
Hi Michael,

Your remark will be transmitted to our development teams to fix this typo. Thank you for having raised the point.

With regards,

ALABSTM, 2019-11-25:
ST Internal Reference: 76401

ALABSTM, 2020-01-28:
Hi Michael,

The typo has been corrected. The new file shall be available in the frame of the STM32CubeF4 **v1.25.0** which will be published soon, hopefully.

Thank you once more for having drawn our attention to such a detail. I think this issue can be closed now.

With regards,

closed 2020-01-28 by ALABSTM