STMicroelectronics/STM32CubeF4 #8: I2S DMA Complete transfer IRQ is not used on circular mode

elgarbe, 2020-02-06:
I'm working with stm32f411re uC and a TI audiocodec through I2S2 interface in full duplex master.
I generate the project using CubeMx 5.5 and HAL driver 1.24.2.
I configure the I2S module to use DMA on both, tx and rx in circular mode. I need to check HalfComplete and Complete transfer in order to prepare/process received data.

But HAL driver prevents Complete callback to be called in circular mode, which is a hug bug. This bug is not present on older versions of the library.
I've reported this one year ago:
https://community.st.com/s/question/0D50X0000AAIIetSQH/i2s-dma-complete-transfer-irq-on-circular-mode

The problem is here
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/7fabee45fb2cbe2189dd40cbe677917448510258/Src/stm32f4xx_hal_i2s_ex.c#L872 :

```
/**
  * @brief  DMA I2S transmit receive process complete callback
  * @param  hdma pointer to a DMA_HandleTypeDef structure that contains
  *               the configuration information for the specified DMA module.
  * @retval None
  */
static void I2SEx_TxRxDMACplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef *hi2s = (I2S_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;

  /* if DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL)
  {
    if (hi2s->hdmarx == hdma)
    {
      /* Disable Rx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_RXDMAEN);
      }
      else
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
      }

      hi2s->RxXferCount = 0U;

      if (hi2s->TxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

        /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }

    if (hi2s->hdmatx == hdma)
    {
      /* Disable Tx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_TXDMAEN);
      }
      else
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_TXDMAEN);
      }

      hi2s->TxXferCount = 0U;

      if (hi2s->RxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

        /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }
  }
}

```

The first check is DMA is configured in NORMAL mode. But that prevents the DMA complete callback to fire an IRQ in CIRCULAR mode, which I can't explain why is that!!!!

elgarbe, 2020-02-06:
In old version of the driver the code was:

```
void I2S_DMATxCplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef* hi2s = (I2S_HandleTypeDef*)((DMA_HandleTypeDef*)hdma)->Parent;
  
  if((hdma->Instance->CR & DMA_SxCR_CIRC) == 0)
  {
    hi2s->TxXferCount = 0;

    /* Disable Tx DMA Request */
    hi2s->Instance->CR2 &= (uint32_t)(~SPI_CR2_TXDMAEN);
#if defined(STM32F427xx) || defined(STM32F437xx) || defined(STM32F429xx) || defined(STM32F439xx) ||\
    defined(STM32F405xx) || defined(STM32F415xx) || defined(STM32F407xx) || defined(STM32F417xx) ||\
    defined(STM32F401xC) || defined(STM32F401xE) || defined(STM32F411xE) || defined(STM32F469xx) ||\
    defined(STM32F479xx)
    if(hi2s->Init.FullDuplexMode == I2S_FULLDUPLEXMODE_ENABLE)
    {
      /* Disable Rx DMA Request for the slave*/  
      I2SxEXT(hi2s->Instance)->CR2 &= (uint32_t)(~SPI_CR2_RXDMAEN);
    }
#endif /* STM32F40xxx || STM32F41xxx || STM32F42xxx || STM32F43xxx || STM32F401xx || STM32F411xx ||\
          STM32F469xx || STM32F479xx */
    if(hi2s->State == HAL_I2S_STATE_BUSY_TX_RX)
    {
      if(hi2s->RxXferCount == 0)
      {
        hi2s->State = HAL_I2S_STATE_READY;
      }
    }
    else
    {
      hi2s->State = HAL_I2S_STATE_READY; 
    }
  }
  HAL_I2S_TxCpltCallback(hi2s);
}
```
So, CpltCallback was called every time.

I've just change current code to this:

```
/**
  * @brief  DMA I2S transmit receive process complete callback
  * @param  hdma pointer to a DMA_HandleTypeDef structure that contains
  *               the configuration information for the specified DMA module.
  * @retval None
  */
static void I2SEx_TxRxDMACplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef *hi2s = (I2S_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;

  /* if DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL)
  {
    if (hi2s->hdmarx == hdma)
    {
      /* Disable Rx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_RXDMAEN);
      }
      else
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
      }

      hi2s->RxXferCount = 0U;

      if (hi2s->TxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

//        /* Call user TxRx complete callback */
//#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
//        hi2s->TxRxCpltCallback(hi2s);
//#else
//        HAL_I2SEx_TxRxCpltCallback(hi2s);
//#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }

    if (hi2s->hdmatx == hdma)
    {
      /* Disable Tx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_TXDMAEN);
      }
      else
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_TXDMAEN);
      }

      hi2s->TxXferCount = 0U;

      if (hi2s->RxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

//        /* Call user TxRx complete callback */
//#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
//        hi2s->TxRxCpltCallback(hi2s);
//#else
//        HAL_I2SEx_TxRxCpltCallback(hi2s);
//#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }
  }
  /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
}
```

and it's working as expected.

Best regards, Leo

ALABSTM, 2020-03-13:
Hi Elgarbe,

As mentioned in the other issue you posted, this point will be forwarded to our development teams for analysis. Thank you again for your contribution.

With regards,

ALABSTM, 2020-03-13:
Hi,

Please consider also this [suggestion](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/issues/1#issuecomment-584820157) as a possible solution to the issue.

With regards,

elgarbe, 2020-03-15:
Hi, that link don't provide a solution to the issue. I already show [here](https://github.com/STMicroelectronics/STM32CubeF4/issues/8#issuecomment-583158208) what is preventing the transfer complete callback to be executed in circular mode, and how to fix it. The user you link to point out another issue, a double callback from transmit and received interrupt.

ALABSTM, 2020-03-16:
Hi Elgarbe,

Thank you for the clarification. I will let the development teams updated. We will also keep an eye on what the other user posted in case it reveals to be another issue as you mentioned.

With regards,

ASELSTM, 2020-04-08:
Hi Elgarbe,

Our development teams confirmed your request. Actually, we already identified this issue since a while, before we receive your request. An internal bug tracker has been already opened. 

The fix will be made available in a future release. We cannot share a date for the moment, unfortunately. We will try to keep you updated. Thank you again.

With regards,

ASELSTM, 2020-04-08:
ST Internal Reference: 51805

ryback08, 2020-08-06:
Thanks elgarbe for your correction. 
HAL_I2SEx_TxRxCpltCallback is now call.

But HAL_I2SEx_TxRxHalfCpltCallback & HAL_I2SEx_TxRxCpltCallback seems to be call twice.
Can you confirm my observation ?
To test this function, I just insert "HAL_GPIO_TogglePin" in Half and Cplt callback custom function.
Normally, we should have a square signal on my output pin.
But I see impulse (with good timing).

Luchao90, 2020-10-29:
Hello, observation from @ryback08  is correct.
With @elgarbe modification interrupt is called twice, one per complete transfer and a little time after per complete receive transfer. This time depends on CLOCK speed.
I suggest to use in normal mode and start DMA into complete interrupt again.
Here you can change buffer if you want.

[image]

ALABSTM, 2020-12-24:
Hi everyone,

An internal fix has been implemented and will be made available in the frame of a future release.

In the meanwhile, below are the details of the fix hoping they help you overcome the problem. The updated `stm32f4xx_hal_i2s_ex.c` file is [attached](https://github.com/STMicroelectronics/STM32CubeF4/files/5739847/STM32F4xx_HAL_Driver.zip). Please let us know you feedback and thank you all for your contributions.

With regards,

```diff
HAL_StatusTypeDef HAL_I2SEx_TransmitReceive_DMA(I2S_HandleTypeDef *hi2s,
                                                uint16_t *pTxData,
                                                uint16_t *pRxData,
                                                uint16_t Size)
{
  uint32_t *tmp = NULL;
  uint32_t tmp1 = 0U;
  HAL_StatusTypeDef errorcode = HAL_OK;
  ...
-  /* Set the I2S Tx DMA Half transfer complete callback */
+  /* Set the I2S Tx DMA Half transfer complete callback as NULL */
-  hi2s->hdmatx->XferHalfCpltCallback  = I2SEx_TxRxDMAHalfCplt;
+  hi2s->hdmatx->XferHalfCpltCallback  = NULL;

-  /* Set the I2S Tx DMA transfer complete callback */
+  /* Set the I2S Tx DMA transfer complete callback as NULL */
-  hi2s->hdmatx->XferCpltCallback  = I2SEx_TxRxDMACplt;
+  hi2s->hdmatx->XferCpltCallback  = NULL;
  ...
}
```

```diff
static void I2SEx_TxRxDMACplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef *hi2s = (I2S_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;

-  /* if DMA is configured in DMA_NORMAL mode */
+  /* If DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL)
  {
-    if (hi2s->hdmarx == hdma)
-    {
-      /* Disable Rx DMA Request */
+    /* Disable Tx & Rx DMA Requests */
    if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
        ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
    {
      CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_RXDMAEN);
-      }
-      else
-      {
-        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
-      }
-
-      hi2s->RxXferCount = 0U;
-
-      if (hi2s->TxXferCount == 0U)
-      {
-        hi2s->State = HAL_I2S_STATE_READY;
-
-        /* Call user TxRx complete callback */
-#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
-        hi2s->TxRxCpltCallback(hi2s);
-#else
-        HAL_I2SEx_TxRxCpltCallback(hi2s);
-#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
-      }
-    }
-
-    if (hi2s->hdmatx == hdma)
-    {
-      /* Disable Tx DMA Request */
-      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
-          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
-      {
      CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_TXDMAEN);
    }
    else
    {
+      CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
      CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_TXDMAEN);
    }

+    hi2s->RxXferCount = 0U;
    hi2s->TxXferCount = 0U;

-      if (hi2s->RxXferCount == 0U)
-      {
    hi2s->State = HAL_I2S_STATE_READY;
+  }
  /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
  hi2s->TxRxCpltCallback(hi2s);
#else
  HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
}
-    }
-  }
-}
```

howard0su, 2021-01-30:
I hit the same issue with F303. I checked the code which looks similar as F4. Can you confirm if they are the same issues?

ASELSTM, 2021-03-03:
Hi @howard0su,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.26.0 of the STM32CubeF4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM