STMicroelectronics/STM32CubeF4 #11: about USBD_SUPPORT_USER_STRING macro definition

ZemingLiu, 2020-03-19:
**USBD_SUPPORT_USER_STRING** was renamed to **USBD_SUPPORT_USER_STRING_DESC** in version **1.25.0**, but STM32CubeMX(v5.6.0) still generates USBD_SUPPORT_USER_STRING in the "usbd_conf.h" file. This will cause some problems, such as the host cannot get the memory map of the target board during DFU.

* In the "usbd_ctlreq.c" file, line 494

```c
#if (USBD_SUPPORT_USER_STRING_DESC == 1U)
      if (pdev->pClass->GetUsrStrDescriptor != NULL)
      {
        pbuf = pdev->pClass->GetUsrStrDescriptor(pdev, (req->wValue), &len);
      }
      else
      {
        USBD_CtlError(pdev, req);
        err++;
      }
#elif (USBD_CLASS_USER_STRING_DESC == 1U)
      if (pdev->pDesc->GetUserStrDescriptor != NULL)
      {
        pbuf = pdev->pDesc->GetUserStrDescriptor(pdev->dev_speed, (req->wValue), &len);
      }
      else
      {
        USBD_CtlError(pdev, req);
        err++;
      }
#else
      USBD_CtlError(pdev, req);
      err++;
#endif
```

* In the "usbd_conf.h" file
```c
/** @defgroup USBD_CONF_Exported_Defines USBD_CONF_Exported_Defines
  * @brief Defines for configuration of the Usb device.
  * @{
  */

/*---------- -----------*/
#define USBD_MAX_NUM_INTERFACES     1U
/*---------- -----------*/
#define USBD_MAX_NUM_CONFIGURATION     1U
/*---------- -----------*/
#define USBD_MAX_STR_DESC_SIZ     512U
/*---------- -----------*/
#define USBD_SUPPORT_USER_STRING     1U
/*---------- -----------*/
#define USBD_DEBUG_LEVEL     0U
/*---------- -----------*/
#define USBD_LPM_ENABLED     1U
/*---------- -----------*/
#define USBD_SELF_POWERED     1U
/*---------- -----------*/
#define USBD_DFU_MAX_ITF_NUM     1U
/*---------- -----------*/
#define USBD_DFU_XFER_SIZE     4096U
/*---------- -----------*/
#define USBD_DFU_APP_DEFAULT_ADD     0x08008000U

/****************************************/
/* #define for FS and HS identification */
#define DEVICE_FS 		0
#define DEVICE_HS 		1

/**
  * @}
  */
```

[image]

Thanks

ALABSTM, 2020-04-02:
Hi @ZemingLiu,

Thank you for this one other issue you reported. However, this one is rather related to **Cube MX** than to the firmware published within this repository. We do apologize, but addressing this kind of issues is out of the scope of our activity.

Please allow me to redirect you to our **ST Community**. You can report this issue in the Cube MX-related [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There, you will hopefully find people from our Cube MX development teams who will take in charge your request.

Thank you for your comprehension.

With regards,

ALABSTM, 2020-04-03:
Hi @ZemingLiu,

Please allow me to introduce @bouattay who could confirm my first analysis (or have a different opinion).

With regards,

ALABSTM, 2020-04-13:
Hi @ZemingLiu,

Our development teams confirmed this is a "Cube MX" related issue. They expect the MW USB Device library v2.6.0 to be supported for STM32F4xx series in "Cube MX" v6.2.0 **or later**. So stay tuned and thank you in advance for your patience.

Please allow me to close this issue now. Thank you again for your report.

With regards,

closed 2020-04-13 by ALABSTM