STMicroelectronics/STM32CubeF4 #13: RCC_CAN1_CLK not available in hal_can.c

Smikha, 2020-03-22:
https://github.com/STMicroelectronics/STM32CubeF4/blob/a86ecaa2fb63029596ba7dabadab2d9c2c139560/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_can.c#L22

Hello, reading the documentation above to initialize the CAN1 interface on my STM32F4, I cannot find the __HAL_RCC_CAN1_CLK_ENABLE() function anywhere in the code. Should this documentation be updated?

Smikha, 2020-03-30:
Hey, I've found the issue to be that this is handled in /Src/stm32f4xx_hal_msp.c within the function of void HAL_CAN_MspInit(CAN_HandleTypeDef* hcan)

Apologies for any confusion. Although I'm still having difficulty getting CAN1 working per your documentation and examples for the STM32F4 boards.

ASELSTM, 2020-03-31:
Hi Smikha,

First of all i want to thank you for your contribution. Glad you were able to find the problem.
Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue.

Otherwise, I invite you to check in the forum (https://community.st.com/s/topic/0TO0X000000BThxWAG/can) you can find more information about the CAN, you may also ask questions about the issue you are facing with the activation of the CAN 1. 

With regards,

closed 2020-03-31 by ASELSTM