STMicroelectronics/STM32CubeF4 #14: UART baudrate and SystemCoreClock incorrect

ChrisIdema, 2020-03-31:
**Describe the set-up**
 * The board: nucleo_f446ze
 * IDE: vscode+platformio+stm32cube framework
* Extra lib: STM32Cube Middleware-FreeRTOS 

**Describe the bug**
I setup uart3 for a baudrate of 115200, but I get a baudrate of 38400.
Also SystemCoreClock= 525000000. Which is clearly wrong.

**How To Reproduce**
1. create a project in stm32cubemx. I setup uart3 and other uarts + USB.

2. generate code using these settings: https://github.com/ussserrr/stm32pio#example

3. copy all the generated source files to lib/stm/src. Test uart printing:
```
uint8_t hewo[]= "Hello world!\r\n";
HAL_UART_Transmit(&huart3,hewo,sizeof(hewo)-1,10000);
vTaskDelay(500);
```

4. I get non-ascii characters at 115200 baud and get Hello world! at 38400 baud. Debugging shows crazy values for pll such as pllvco=1050000000 and SystemCoreClock= 525000000. The parameters seem to match the clock configuration.

Contents of my .ioc-file:
```
#MicroXplorer Configuration settings - do not modify
File.Version=6
KeepUserPlacement=false
Mcu.Family=STM32F4
Mcu.IP0=NVIC
Mcu.IP1=RCC
Mcu.IP10=USB_OTG_FS
Mcu.IP2=SYS
Mcu.IP3=UART4
Mcu.IP4=UART5
Mcu.IP5=USART1
Mcu.IP6=USART2
Mcu.IP7=USART3
Mcu.IP8=USART6
Mcu.IP9=USB_DEVICE
Mcu.IPNb=11
Mcu.Name=STM32F446Z(C-E)Tx
Mcu.Package=LQFP144
Mcu.Pin0=PC13
Mcu.Pin1=PC14-OSC32_IN
Mcu.Pin10=PD8
Mcu.Pin11=PD9
Mcu.Pin12=PG6
Mcu.Pin13=PG7
Mcu.Pin14=PC6
Mcu.Pin15=PA8
Mcu.Pin16=PA9
Mcu.Pin17=PA10
Mcu.Pin18=PA11
Mcu.Pin19=PA12
Mcu.Pin2=PC15-OSC32_OUT
Mcu.Pin20=PA13
Mcu.Pin21=PA14
Mcu.Pin22=PB6
Mcu.Pin23=PB7
Mcu.Pin24=VP_SYS_VS_Systick
Mcu.Pin25=VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS
Mcu.Pin3=PH0-OSC_IN
Mcu.Pin4=PH1-OSC_OUT
Mcu.Pin5=PA0-WKUP
Mcu.Pin6=PA2
Mcu.Pin7=PB0
Mcu.Pin8=PE8
Mcu.Pin9=PB14
Mcu.PinsNb=26
[... 231 lines cut ...]
USART1.IPParameters=VirtualMode,BaudRate
USART1.VirtualMode=VM_ASYNC
USART2.BaudRate=2000000
USART2.IPParameters=VirtualMode,BaudRate
USART2.VirtualMode=VM_ASYNC
USART3.IPParameters=VirtualMode
USART3.VirtualMode=VM_ASYNC
USART6.BaudRate=2000000
USART6.IPParameters=VirtualMode,BaudRate
USART6.VirtualMode=VM_ASYNC
USB_DEVICE.CLASS_NAME_FS=CDC
USB_DEVICE.IPParameters=VirtualMode-CDC_FS,VirtualModeFS,CLASS_NAME_FS,MANUFACTURER_STRING-CDC_FS,PRODUCT_STRING_CDC_FS
USB_DEVICE.MANUFACTURER_STRING-CDC_FS=Electron Square
USB_DEVICE.PRODUCT_STRING_CDC_FS=Electron Square tty
USB_DEVICE.VirtualMode-CDC_FS=Cdc
USB_DEVICE.VirtualModeFS=Cdc_FS
USB_OTG_FS.IPParameters=VirtualMode
USB_OTG_FS.VirtualMode=Device_Only
VP_SYS_VS_Systick.Mode=SysTick
VP_SYS_VS_Systick.Signal=SYS_VS_Systick
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Mode=CDC_FS
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Signal=USB_DEVICE_VS_USB_DEVICE_CDC_FS
board=NUCLEO-F446ZE
boardIOC=true
```

ChrisIdema, 2020-04-01:
I've found a work around. If I select HSI instead of HSE as a clock source it works. The only problem is that STM32CubeMX says it cannot resolve the clock issues in that case, but it still works if I ignore the error message and proceed to generate code.
HSE is not connected to a crystal, but to an ST-link output. That could be an issue.

Here is my new ioc file:

```
#MicroXplorer Configuration settings - do not modify
FREERTOS.IPParameters=Tasks01
FREERTOS.Tasks01=defaultTask,24,128,StartDefaultTask,Default,NULL,Dynamic,NULL,NULL
File.Version=6
GPIO.groupedBy=Group By Peripherals
KeepUserPlacement=false
Mcu.Family=STM32F4
Mcu.IP0=FREERTOS
Mcu.IP1=NVIC
Mcu.IP2=RCC
Mcu.IP3=RTC
Mcu.IP4=SYS
Mcu.IP5=UART4
Mcu.IP6=UART5
Mcu.IP7=USART3
Mcu.IP8=USB_DEVICE
Mcu.IP9=USB_OTG_FS
Mcu.IPNb=10
Mcu.Name=STM32F446Z(C-E)Tx
Mcu.Package=LQFP144
Mcu.Pin0=PC13
Mcu.Pin1=PC14-OSC32_IN
Mcu.Pin10=PD9
Mcu.Pin11=PG6
Mcu.Pin12=PG7
Mcu.Pin13=PA8
Mcu.Pin14=PA9
Mcu.Pin15=PA10
Mcu.Pin16=PA11
Mcu.Pin17=PA12
Mcu.Pin18=PA13
Mcu.Pin19=PA14
Mcu.Pin2=PC15-OSC32_OUT
Mcu.Pin20=PB7
Mcu.Pin21=VP_FREERTOS_VS_CMSIS_V2
Mcu.Pin22=VP_RTC_VS_RTC_Activate
Mcu.Pin23=VP_RTC_VS_RTC_Calendar
Mcu.Pin24=VP_SYS_VS_tim1
Mcu.Pin25=VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS
Mcu.Pin3=PH0-OSC_IN
Mcu.Pin4=PH1-OSC_OUT
Mcu.Pin5=PA0-WKUP
Mcu.Pin6=PB0
Mcu.Pin7=PE8
Mcu.Pin8=PB14
Mcu.Pin9=PD8
Mcu.PinsNb=26
Mcu.ThirdPartyNb=0
Mcu.UserConstants=
Mcu.UserName=STM32F446ZETx
MxCube.Version=5.6.0
MxDb.Version=DB.5.0.60
NVIC.BusFault_IRQn=true\:0\:0\:false\:false\:true\:false\:true\:false
NVIC.DebugMonitor_IRQn=true\:0\:0\:false\:false\:true\:false\:true\:false
NVIC.ForceEnableDMAVector=true
NVIC.HardFault_IRQn=true\:0\:0\:false\:false\:true\:false\:true\:false
NVIC.MemoryManagement_IRQn=true\:0\:0\:false\:false\:true\:false\:true\:false
NVIC.NonMaskableInt_IRQn=true\:0\:0\:false\:false\:true\:false\:true\:false
NVIC.OTG_FS_IRQn=true\:5\:0\:false\:false\:true\:true\:false\:true
NVIC.PendSV_IRQn=true\:15\:0\:false\:false\:false\:true\:true\:false
NVIC.PriorityGroup=NVIC_PRIORITYGROUP_4
NVIC.SVCall_IRQn=true\:0\:0\:false\:false\:false\:false\:true\:false
NVIC.SysTick_IRQn=true\:15\:0\:false\:false\:false\:true\:true\:false
NVIC.TIM1_UP_TIM10_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:true
[... 206 lines cut ...]
UART5.BaudRate=2000000
UART5.IPParameters=VirtualMode,BaudRate
UART5.VirtualMode=Asynchronous
USART3.IPParameters=VirtualMode
USART3.VirtualMode=VM_ASYNC
USB_DEVICE.CLASS_NAME_FS=CDC
USB_DEVICE.IPParameters=VirtualMode-CDC_FS,VirtualModeFS,CLASS_NAME_FS
USB_DEVICE.VirtualMode-CDC_FS=Cdc
USB_DEVICE.VirtualModeFS=Cdc_FS
USB_OTG_FS.IPParameters=VirtualMode
USB_OTG_FS.VirtualMode=Device_Only
VP_FREERTOS_VS_CMSIS_V2.Mode=CMSIS_V2
VP_FREERTOS_VS_CMSIS_V2.Signal=FREERTOS_VS_CMSIS_V2
VP_RTC_VS_RTC_Activate.Mode=RTC_Enabled
VP_RTC_VS_RTC_Activate.Signal=RTC_VS_RTC_Activate
VP_RTC_VS_RTC_Calendar.Mode=RTC_Calendar
VP_RTC_VS_RTC_Calendar.Signal=RTC_VS_RTC_Calendar
VP_SYS_VS_tim1.Mode=TIM1
VP_SYS_VS_tim1.Signal=SYS_VS_tim1
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Mode=CDC_FS
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Signal=USB_DEVICE_VS_USB_DEVICE_CDC_FS
board=NUCLEO-F446ZE
boardIOC=true

```

ASELSTM, 2020-04-02:
Hi @ChrisIdema 

First, allow me to thank you for your contribution. However, aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at ST Community level, precisely within the [Cube MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section.

There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit your issue.

Please, allow me to close this issue. Thank you for your comprehension.

Thank you,

closed 2020-04-02 by ASELSTM