STMicroelectronics/STM32CubeF4 #15: STM32F407 Discovery working as Slave Transmiter not correct.

VoLinhTruc, 2020-05-16:
I test that the STM32F407 Discovery working as receiver is ok (no any transmit function like HAL_SPI_Transmit or HAL_SPI_TransmitReceive).
But after I added the transmit function for the purpose to send data to Master, it does not work correctly.
I described the issue in the video below.
Hope to receive your help.
Thank you so much.

https://www.youtube.com/watch?v=xA6N_trzkb0

ASELSTM, 2020-05-26:
Hi @VoLinhTruc,

Thank you for your contribution. 

Unfortunately, we can't access to the video as it's private. In order to allow to our development teams to get a better understanding of the issue, would you please make the video public so that they can easily access to it.

With regards,

ASELSTM, 2020-07-30:
Hi @VoLinhTruc,

Any feedback about this issue to move this discussion forward ?

Thank you,

ASELSTM, 2020-08-17:
Hi @VoLinhTruc,

Thank you once more for your contribution. However, without more details we won't be able to investigate this issue further.
So, please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension.

With regards,

closed 2020-08-17 by ASELSTM