STMicroelectronics/STM32CubeF4 #16: STemWin MIPI DSI driver in video mode

m0nka, 2020-05-17:
Hello,

I was digging through old releases of the CubeF4, it seems that the original example in:

Projects\STM32469I-Discovery\Applications\STemWin\STemWin_HelloWorld

was using the LCD in video mode (1.9.0 release), then at some point this was abandoned in
the favor of AdaptedCommandMode, which is the current example in 1.25.0.

Off course this is not visible in the History of the LCDConf_stm32469i_discovery.c as the initial
upload of this repo was based on much later release. There are lots of examples that use Video
mode, but those use basic (bare metal) gui libs, not anything more advanced like STemWin.

Do you guys have another more recent and working example of this driver somewhere please ?
As MIPI DSI LCDs with internal GRAM are rare as unicorns, every one that could be bought and 
used on a project needs video mode and has no internal RAM. Thank you!

Regards, Krassi

ALABSTM, 2020-06-08:
Hi @m0nka,

I hope you are doing well. You request will be forwarded to our development teams. I will let you updated once they provide me with their answer.

With regards,

ALABSTM, 2021-08-02:
Hi @m0nka,

I hope you are fine. I do not know whether you could find a way to cope with your request. Otherwise, I can suggest you one thanks to our development teams. Using your **Cube MX** tool, you can download version **v1.9.0** of the STM32CubeF4 firmware package.

1. From the main screen of Cube MX, drop down menu `Help` and select option `Manage embedded software package`.
2. In the pop-up window, scroll down the list of packages till the `STM32F4` entry.
3. Drop down the list then scroll down till version **1.9.0**.

Below, a couple of screenshots illustrating each step. I hope this helps. Thank you for your patience and please excuse this late reply.

With regards,

[image]

[image]

[image]

ALABSTM, 2021-08-05:
Hi @m0nka,

I hope my previous comment is helpful. Please allow me to close this issue now.

With regards,

closed 2021-08-05 by ALABSTM

m0nka, 2021-08-06:
Great, thank you!