STMicroelectronics/STM32CubeF4 #17: stm32f4xx_hal_rcc.c  line:545  check RCC_PLLCFGR_PLLP  registers error

CAT19TH, 2020-05-18:
Hi Community,

I find a bug in STM32CubeF4 Ver 1.25.0  stm32f4xx_hal_rcc.c  line 546.

This wil return HAL_ERROR when the register is correct.

>​        /* Do not return HAL_ERROR if request repeats the current configuration */
>        pll_config = RCC->PLLCFGR;
>        if((READ_BIT(pll_config, RCC_PLLCFGR_PLLSRC) != RCC_OscInitStruct->PLL.PLLSource) ||
>           (READ_BIT(pll_config, RCC_PLLCFGR_PLLM) != RCC_OscInitStruct->PLL.PLLM) ||
>           (READ_BIT(pll_config, RCC_PLLCFGR_PLLN) != RCC_OscInitStruct->PLL.PLLN) ||
>           (READ_BIT(pll_config, RCC_PLLCFGR_PLLP) != RCC_OscInitStruct->PLL.PLLP) ||
>           (READ_BIT(pll_config, RCC_PLLCFGR_PLLQ) != RCC_OscInitStruct->PLL.PLLQ))
>        {
>          return HAL_ERROR;
>        }  

​base on stm32f4xx_hal_rcc.c  line 497:

>​        /* Configure the main PLL clock source, multiplication and division factors. */
>        WRITE_REG(RCC->PLLCFGR, (RCC_OscInitStruct->PLL.PLLSource                                            | \
>                                 RCC_OscInitStruct->PLL.PLLM                                                 | \
>                                 (RCC_OscInitStruct->PLL.PLLN << RCC_PLLCFGR_PLLN_Pos)             | \
>                                 (((RCC_OscInitStruct->PLL.PLLP >> 1U) - 1U) << RCC_PLLCFGR_PLLP_Pos) | \
>                                 (RCC_OscInitStruct->PLL.PLLQ << RCC_PLLCFGR_PLLQ_Pos)));

and stm32f4xx_hal_rcc.h  line 497:

>/** @defgroup RCC_PLLP_Clock_Divider PLLP Clock Divider
>  * @{
>  */
>#define RCC_PLLP_DIV2                  0x00000002U
>#define RCC_PLLP_DIV4                  0x00000004U
>#define RCC_PLLP_DIV6                  0x00000006U
>#define RCC_PLLP_DIV8                  0x00000008U

and RM009 page 164:
​
>Bits 17:16 PLLP: Main PLL (PLL) division factor for main system clock
>Set and cleared by software to control the frequency of the general PLL output clock. These
>bits can be written only if PLL is disabled.
>Caution: The software has to set these bits correctly not to exceed 180 MHz on this domain.
>PLL output clock frequency = VCO frequency / PLLP with PLLP = 2, 4, 6, or 8
>>00: PLLP = 2
>>01: PLLP = 4
>>10: PLLP = 6
>>11: PLLP = 8

stm32f4xx_hal_rcc.c line 546: 
>(READ_BIT(pll_config, RCC_PLLCFGR_PLLP) != RCC_OscInitStruct->PLL.PLLP) ||  

should be replace by :  
>(READ_BIT(pll_config, RCC_PLLCFGR_PLLP) != ((RCC_OscInitStruct->PLL.PLLP >> 1U) - 1U)) ||

CAT19TH, 2020-05-19:
stm32f4xx_hal_rcc.c begin in line 541 ,should be replace by :
>
        /* Do not return HAL_ERROR if request repeats the current configuration */  
        pll_config = RCC->PLLCFGR;  
        if ((READ_BIT(pll_config, RCC_PLLCFGR_PLLSRC) != RCC_OscInitStruct->PLL.PLLSource) ||  
            (READ_BIT(pll_config, RCC_PLLCFGR_PLLM) >> RCC_PLLCFGR_PLLM_Pos != RCC_OscInitStruct->PLL.PLLM) ||  
            (READ_BIT(pll_config, RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos != RCC_OscInitStruct->PLL.PLLN) ||  
            (READ_BIT(pll_config, RCC_PLLCFGR_PLLP) >> RCC_PLLCFGR_PLLP_Pos != ((RCC_OscInitStruct->PLL.PLLP >> 1U) - 1U)) ||  
            (READ_BIT(pll_config, RCC_PLLCFGR_PLLQ) >> RCC_PLLCFGR_PLLQ_Pos != RCC_OscInitStruct->PLL.PLLQ))  
        {  
          return HAL_ERROR;  
        }

ASELSTM, 2020-05-19:
Hi @CAT19TH,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-05-19:
ST Internal Reference: 86633

rleigh-lumiradx, 2020-12-14:
@ASELSTM Is there a timeline for this fix?

@CAT19TH I think your fix is missing the shift of `RCC_PLLCFGR_PLLSRC_Pos` by `RCC_PLLCFGR_PLLSRC_Pos`.  But it's likely zero so this was missed.

If that's been missed in the internal tracker/source change, please could you make sure it hasn't been missed as well.

rleigh-lumiradx, 2020-12-14:
By the way, this broken code is being used by default when you start a new project with CubeMX.  I just created a new one for the Nucleo-F429ZI, and encountered the bug this way.  Following some debugging, I found the logic for the RCC was incorrect as shown above.   Please bear in mind that it's not currently possible to generate a working project for any F4 boards using CubeMX or via direct use of your GitHub releases.

ASELSTM, 2020-12-14:
Hi @rleigh-lumiradx,

The fix has been implemented internally and will be published in the frame of a next release. Unfortunately, we cannot share an exact date of our next STM32CubeF4 release for the moment. We will keep you informed.

With regards,

theadib, 2021-01-12:
The problem is still there in STM32F4 1.25.2.
It costed my 2hours to spot.
Thanks again for not fixing bugs.

ASELSTM, 2021-01-15:
Hi @theadib, 

Actually, as long as the issue is not yet closed or its status is still "**To release**" it means that the fix is not yet available in GitHub.  We will keep you informed once the fix is available on GitHub, stay tuned. 

You may also refer to the [**Release note**](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Release_Notes.html) to check the main changes.

With regards,

ASELSTM, 2021-03-03:
Hi @CAT19TH,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.26.0 of the STM32CubeF4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM

InconsolableCellist, 2021-07-28:
I believe this is still broken. In `stm32f4xx_hal_rcc.c` on line 553 the `HEAD` in `master` from git shows:

  `          (READ_BIT(pll_config, RCC_PLLCFGR_PLLSRC) != RCC_OscInitStruct->PLL.PLLSource) `

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c#L553

There was a partial fix in `d599a824d`, but you need to also shift the PLL.PLLSource over by `RCC_PLLCFGR_PLLSRC_Pos` like the other ones, or perhaps better, use `READ_BIT`.

I believe this error may be repeated on lines 545 and 1043, but I didn't check those codepaths.

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c#L545

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c#L1043

----

Additionally, there seems to be a regression introduced by `55c4943bb8` where it returns HAL_ERROR despite a valid PLL config, generated by STM32CubeMX 6.3.0 on the STM32F407 DISC-1 board using the following (and apparently valid) clock configuration: https://i.imgur.com/Bz2ifz1.png

In the latest git version of the code (2f3b26f1, `master` branch) there's a discrepancy between `RCC->PLLCFGR_PLLN` and the requested `RCC_OscInitStruct->PLL.PLLN`. In this case, I'm requesting the valid PLLN of 336 (RM0090 Rev. 19, p. 227), but the old PLLN is something different. The `RCC->PLLCFGR` is never updated by anything (like it is on line 78, but only in the case of it not being a system clock already).

I think the fundamental issue may be that STM32CubeMX is generating init code for using a PLL as a system clock, while the safety checks in that HAL file assume you're on a non-PLL and trying to switch to it--and it only lets you if your configuration is identical to what it was before.

InconsolableCellist, 2021-07-28:
@ASELSTM please see my above comment and reopen ^

InconsolableCellist, 2021-07-28:
For anyone discovering this issue via a search engine, the workaround I'm using is to let STM32CubeMX configure everything for the HSI, then add a new function to switchover to the PLL. "Mastering STM32" is useful for learning how with the HAL. My function looks like this, but there may be bugs, I'm new to these chips: 

```
void SystemClock_SwitchToPLL() {
    RCC_OscInitTypeDef RCC_OscInitStruct = {0};
    RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
    RCC_OscInitStruct.PLL.PLLM = 8;
    RCC_OscInitStruct.PLL.PLLN = 336;
    RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
    RCC_OscInitStruct.PLL.PLLQ = 4;
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
        Error_Handler();
    }

    RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_SYSCLK;
    RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
    RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
    RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
    RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
    if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
        Error_Handler();
    }

    HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);
    HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);
    HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}
```

I'm pretty sure the NVIC and SYSTICK calls are handled by the generated MSP already, but I don't think it hurts to have here.

xor-shift, 2023-01-08:
Sorry for the necro but I would like to leave a note for those stumbling upon this issue (or #84 for that matter) through search engines:
I have encountered the issue described here only while debugging with OpenOCD on CLion and after narrowing down my searches, I came across [this EEVBlog thread](https://www.eevblog.com/forum/microcontrollers/stm32-clock-gets-modified-when-debugger-is-connected/msg3306944/#msg3306944) which contains a solution to the issue of OpenOCD messing with the PLL settings.
The biggest hint to the debugger being attached having an effect on the behaviour was Linux being unable to enumerate the STM's OTG device while debugging but enumerating it fine while not debugging.