STMicroelectronics/STM32CubeF4 #20: Parenthesis around if body

the78mole, 2020-05-21:
When macro is used within an if-else statement, it can lead to ambiguities, when not sourrounded by curly braces.

https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/portmacro.h#L92

A better style would be to write:
```
#define portEND_SWITCHING_ISR( xSwitchRequired ) \
    do \
        { if( xSwitchRequired != pdFALSE ) portYIELD(); \
    } while(0)
```

RKOUSTM, 2020-05-27:
Hi @the78mole,

Thank you for this report. Unfortunately we don't treat aspect related to FreeRTOS in our GitHub repositories. As the Third_Party files are out of the scope of STMicroelectronics what about reporting this issue to the [FreeRTOS GitHub](https://github.com/freertos)?  
 
Please allow me to close this issue now. Thank you again for your contribution. 

With regards,

closed 2020-05-27 by RKOUSTM

the78mole, 2020-05-29:
OK, I'll do. Thanks!