STMicroelectronics/STM32CubeF4 #24: I found issue in stm32f4xx_hal_i2c.c

VoLinhTruc, 2020-05-23:
Please check this in the file stm32f4xx_hal_i2c.c

`    if ((hi2c->XferCount == 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX_LISTEN))
    {
      /* Last Byte is received, disable Interrupt */
      __HAL_I2C_DISABLE_IT(hi2c, I2C_IT_BUF);

      /* Set state at HAL_I2C_STATE_LISTEN */
      hi2c->PreviousState = I2C_STATE_SLAVE_BUSY_TX;
      hi2c->State = HAL_I2C_STATE_LISTEN;

      /* Call the corresponding callback to inform upper layer of End of Transfer */
#if (USE_HAL_I2C_REGISTER_CALLBACKS == 1)
      hi2c->SlaveTxCpltCallback(hi2c);
#else
      HAL_I2C_SlaveTxCpltCallback(hi2c);
#endif /* USE_HAL_I2C_REGISTER_CALLBACKS */
    }`

`if ((hi2c->XferCount == 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX_LISTEN))`
must become
`if ((hi2c->XferCount == 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX))`

And
`hi2c->State = HAL_I2C_STATE_LISTEN;`
must become
`hi2c->State = HAL_I2C_STATE_READY;`

ASELSTM, 2020-05-26:
Hi @VoLinhTruc,

Tank you for this new issue you have raised out.

In order to allow a better understanding of the problem, could you please describe how you did to detect this issue and how to reproduce it, so that our development teams can give you an answer. 

With regards,

ASELSTM, 2020-06-12:
Hi @VoLinhTruc,

Any feedback about this issue?

Thank you,

VoLinhTruc, 2020-06-14:
Sorry, recently I'm quite busy so I can not create a video to show a issue. I'll make it some day.
But this is what I met.

1. The conditional stament
`if ((hi2c->XferCount == 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX_LISTEN))`
could not be execute, becaure always false.
This is the reason:
I use this function (`HAL_I2C_Slave_Receive_IT`) to let the I2C turn to receive IT mode.
That function set the state to HAL_I2C_STATE_BUSY_RX, not HAL_I2C_STATE_BUSY_TX_**LISTEN**
`hi2c->State     = HAL_I2C_STATE_BUSY_RX;`

2. 
After the Slave transmited all the data, the state must be HAL_I2C_STATE_READY, not HAL_I2C_STATE_LISTEN.
Because the Slave can not transmit data anymore if the hi2c->State is HAL_I2C_STATE_LISTEN.

JRLSTM, 2020-06-19:
Hi,

Evenif you shared some information, it is not easy to understood your needs.

On your last message you said notion of RX frame with HAL active handle TX status. It is not a normal usecase, mean can you help to find the SW schedule that you used on your side ?

Is it possible for you to shared your software sequence where your detect a mismatch on HAL state for your needs ?

Thanks and regards

ASELSTM, 2020-07-30:
Hi @VoLinhTruc,

I hope you are fine. Any new feedback regarding this issue to move this discussion forward?

With regards,

ASELSTM, 2020-09-16:
Hi @VoLinhTruc,

Thank you once more for your contribution. However, without more details we won't be able to investigate this issue further.
So, please allow me to close this thread as no activity has been registered since a couple of month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension.

With regards

closed 2020-09-16 by ASELSTM

VoLinhTruc, 2020-09-16:
Thank you so much.
I'm so busy recently, so I don't have time to make more detail to you.
I'll provide to you more detail later.

Again, thank you so much for responding