STMicroelectronics/STM32CubeF4 #25: HAL_Delay_us (Feature Request)

jrahlf, 2020-06-02:
Hi,

can you please add a function to create microsecond delays to the HAL, I think this would be greatly appreciated by many users.
On the internet there are implementations using the DWT module, which seems most suitable.

This likely affects not only the F4 variant, but all other STM32 families as well.

Thanks.

SL-RU, 2020-06-04:
It will require additional timer and it is task defined application. 
There is no need in such thing, you can easily modify timer prescaler and few variables to HAL_Delay be 1us.

ALABSTM, 2020-07-22:
Hi @jrahlf,

I hope the links I provided you with were helpful. Please allow me to close this issue now.

With regards,

closed 2020-07-22 by ALABSTM