STMicroelectronics/STM32CubeF4 #26: BSP_AUDIO_IN_Record

laterps, 2020-06-07:
At stm32412g_discovery_audio.c
` if(HAL_OK != HAL_DFSDM_FilterRegularStart_DMA(&hAudioInDfsdmFilter[POS_VAL(INPUT_DEVICE_DIGITAL_MIC1)], pScratchBuff[POS_VAL(INPUT_DEVICE_DIGITAL_MIC1)], ScratchSize))
    {
      return AUDIO_ERROR;
    }`
retuns this error.
I am working with stm32f412-discovery board

ASELSTM, 2020-06-16:
Hi @laterps,

Thank you for your contribution.

We have been trying to reproduce the issue, but we didn't get the behavior described above.

For deeper analysis and in order to better understand the origin of your issue, may i ask you some questions :

1. Would it possible to give us more clarification about this issue 

2. Could you please try to run only the "[AudioRecDfsdm_demo](https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Projects/STM32F412G-Discovery/Examples/BSP/Src/main.c#L78)" by commenting the rest of the demos listed in [BSP_examples ](https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Projects/STM32F412G-Discovery/Examples/BSP/Src/main.c#L65) and checkout if you still have the same issue.

With regards,

ASELSTM, 2020-06-25:
Hi @laterps,

Any feedback about this issue?

Thank you,

ASELSTM, 2020-08-17:
Hi @laterps,

Thank you once more for your contribution. However, without more details we won't be able to investigate this issue further. 
So, please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension.

With regards,

closed 2020-08-17 by ASELSTM