STMicroelectronics/STM32CubeF4 #27: register keyword

wmmc88, 2020-06-11:
**Describe the set-up**
 * stm32f446re
 * arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

**Describe the bug**
`warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]`

**How To Reproduce**
1. Compile with -std=gnu++2a or -std=gnu++17

**Additional context**
Ther warnings show in several different files including `stm32f4xx_ll_usart.h`, `stm32f4xx_ll_tim.h`, `stm32f4xx_ll_adc.h`, `stm32f4xx_ll_rtc.h`, etc. I believe the register keyword should just be removed as its been deprecated since C++11, removed since C++17 and compilers havent used it for a long long time. Would u accept a pr removing use of this keyword?

ARM also removed the use of the "register" keyword in cmsis:
https://github.com/ARM-software/CMSIS_5/commit/f46ec37262bc4fd82b6c91d24198cc3b4a499fe3
https://github.com/ARM-software/CMSIS_5/issues/345

Relevant issue: 
https://github.com/ARMmbed/mbed-os/issues/12232

ASELSTM, 2020-06-12:
Hi @wmmc88,

First of all we would like to thank you for your contribution.

Actually, this issue has already been reported within our GitHub repositories, we have already spotted the problem and fixed it internally. However, the fix has not been published yet. It is scheduled for next release.  

With regards,

ASELSTM, 2020-06-12:
ST Internal Reference: 84396

wmmc88, 2020-06-14:
> Hi @wmmc88,
> 
> First of all we would like to thank you for your contribution.
> 
> Actually, this issue has already been reported within our GitHub repositories ([STM32CubeH7/#15](https://github.com/STMicroelectronics/STM32CubeH7/issues/15)), we have already spotted the problem and fixed it internally. However, the fix has not been published yet. It is scheduled for next release.
> 
> With regards,

Is there a release schedule that you guys follow or do you know when the next release will be out?

ASELSTM, 2020-06-16:
Hi @wmmc88,

Unfortunately, we cannot share an exact date of our next STM32CubeF4 release for the moment. We will keep you informed.

With regards,

ASELSTM, 2021-03-03:
Hi @wmmc88,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.26.0 of the STM32CubeF4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM