STMicroelectronics/STM32CubeF4 #29: compile error for lwip module on Keil project using ARMCC6 compiler

theadib, 2020-06-15:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
I am using STM32 on win32 and Keil-MDK and auto generated project using CubeMX 5.6.1. and STM32F4-1.25.
This comes with lwip 2.1.2

Using the ARM Keil compiler version 6 ARM-CC 6.14 clang based compiler, I run into compiler problem.

**Describe the bug**

../Middlewares/Third_Party/LwIP/system\arch/cc.h(46): error: 'sys/time.h' file not found
#include <sys/time.h>
         ^~~~~~~~~~~~

in file cc.h line 44ff
...
    >#if defined (_GNUC_) & !defined (__CC_ARM)
    >
    >#define LWIP_TIMEVAL_PRIVATE 0
    >#include <sys/time.h>
    >
    >#endif

... 

**How To Reproduce**
1. Indicate the global behavior of your application project.
switch in the Keil uVision IDE compiler generation 5 to compiler generation 6 (this is clang based)
compile using Keil uvision IDE

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
LWIP module does not compile

3. The use case that generates the problem.
using the default compiler with current ARM development set

4. How we can reproduce the problem.
follow 1)

**Additional context**

The test for compiler is not correct (i assume).
There are other tests in cmsis_compiler.h 33ff
...

    >/*
    > * Arm Compiler 4/5
    > */
    >#if   defined ( __CC_ARM )
    >  #include "cmsis_armcc.h"
    >
    >
    >/*
    > * Arm Compiler 6 (armclang)
    > */
    >#elif defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
    >  #include "cmsis_armclang.h"
    >
    >
    >/*
    > * GNU Compiler
    > */
    >#elif defined ( _GNUC_ )
    >  #include "cmsis_gcc.h"

..

I can resolve this error using following patch which applies the test logic of cmsis_compiler.h on that in cc.h

Index: cc.h
===================================================================
--- cc.h (revision 18447)
+++ cc.h (working copy)
@@ -40,7 +40,7 @@

 #define LWIP_PROVIDE_ERRNO

-#if defined (_GNUC_) & !defined (__CC_ARM)
+#if defined (_GNUC_) && !defined (__CC_ARM) && !(defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))

 #define LWIP_TIMEVAL_PRIVATE 0
 #include <sys/time.h> 

I checked the lwip git repository.
There is no such file for STM32 or cortexM controller.
So I think this is not up to lwip to fix but rather STM must fix this bug.

Hoever there is no clear ownership of this file .

regards, Adib.
--

**Screenshots**
If applicable, add screenshots to help explain your problem.

theadib, 2020-06-20:
another instance on lwip.c for supporting sio feature.

Index: lwip.c
===================================================================
--- lwip.c	(revision 18468)
+++ lwip.c	(working copy)
@@ -21,7 +21,7 @@
 #include "lwip.h"
 #include "lwip/init.h"
 #include "lwip/netif.h"
-#if defined ( __CC_ARM )  /* MDK ARM Compiler */
+#if defined ( __CC_ARM ) || (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))  /* MDK ARM Compiler */
 #include "lwip/sio.h"
 #endif /* MDK ARM Compiler */
 #include "ethernetif.h"
@@ -188,7 +188,7 @@
 /* USER CODE END 4_3 */
 }
 
-#if defined ( __CC_ARM )  /* MDK ARM Compiler */
+#if defined ( __CC_ARM )  || (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))/* MDK ARM Compiler */
 /**
  * Opens a serial device for communication.
  *

ALABSTM, 2020-07-21:
Hi @theadib,

Thank you for this report. It will be forwarded to our development teams for analysis. May I however ask you whether the `lwip.c` file is a Cube MX generated one?

With regards,

theadib, 2020-07-21:
Hello Ali, yes lwip.c is generated by cubemx. Regards, Adib.

ALABSTM, 2020-07-21:
Hello @theadib,

Thank you for this quick reply. Your report has already been submitted to our development teams. I will get back to you as soon as they provide me with feedback.

With regards,

ALABSTM, 2020-08-05:
ST Internal Reference: 90928

ALABSTM, 2020-08-05:
Hi @theadib,

The issue you reported has been confirmed by our development teams. The adopted solution is the one you suggested, **i.e.** to update the line below:
https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Middlewares/Third_Party/LwIP/system/arch/cc.h#L43
As follows:
```diff
-#if defined (__GNUC__) & !defined (__CC_ARM)
+#if defined (__GNUC__) && !defined (__CC_ARM) && !(defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))
```
The fix will be made available in a future release. Thank you for you patience and thank you again for your contribution.

With regards,

closed 2021-03-08 by ALABSTM

theadib, 2022-09-30:
I recreated a project using CubeMX 6.6.1 and STM32F4 libs 1.27.1 and this bug is still persisting.

I made a pull request for you to fix it:
https://github.com/STMicroelectronics/stm32_mw_lwip/pull/3

theadib

Sasha7b9, 2025-02-18:
The error is still present in STM32Cube_FW_F4_V1.28.1