STMicroelectronics/STM32CubeF4 #30: template file system_stm32f4xx.c should support userdefined location of vectortable

theadib, 2020-06-23:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.
Project generated by CubeMX

**Describe the bug**
A clear and concise description of what the bug is.
template file system_stm32f4xx.c should support userdefined location of vectortable
using a custom bootloader requires to move the default location of the vector table.
in the file system_stm32f4xx.c that is copied by the program generator is the needed settings for the move of the vector table.
in the defines section at the beginning of the file line 92ff
the lines:
/* #define VECT_TAB_SRAM */
#define VECT_TAB_OFFSET  0x00 /*!< Vector Table base offset field. 
                                   This value must be a multiple of 0x200. */
/******************************************************************************/

have to be changed to
/* #define VECT_TAB_SRAM */
#if !defined(VECT_TAB_OFFSET)
#define VECT_TAB_OFFSET  0x00 /*!< Vector Table base offset field. 
                                   This value must be a multiple of 0x200. */
#endif
/******************************************************************************/

So the user can control the define from compiler command file.
This allows targets for bootloader or plain memory layout.

All the existing code will not break.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RKOUSTM, 2020-07-01:
Hi @theadib,

Thank you for this report. You request will be forwarded to our development teams. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

RKOUSTM, 2020-08-18:
Hi @theadib,

Thank you for your contribution. Our development teams confirmed the issue you pointed out. The fix below will be available in the next release. 
```diff
+ #if defined(USER_VECT_TAB_ADDRESS)
/* #define VECT_TAB_SRAM */
+ #if defined(VECT_TAB_SRAM)
+ #define VECT_TAB_BASE_ADDRESS   SRAM_BASE       /*!< Vector Table base address field.
+                                                    This value must be a multiple of 0x200. */
+ #define VECT_TAB_OFFSET         0x00000000U     /*!< Vector Table base offset field.
+                                                    This value must be a multiple of 0x200. */
+ #else
+ #define VECT_TAB_BASE_ADDRESS   FLASH_BASE      /*!< Vector Table base address field.
+                                                     This value must be a multiple of 0x200. */
+ #define VECT_TAB_OFFSET         0x00000000U     /*!< Vector Table base offset field.
+                                                    This value must be a multiple of 0x200. */
+ #endif /* VECT_TAB_SRAM */
+ #endif /* USER_VECT_TAB_ADDRESS */
```

 I hope this answer helps. If you have no other question about this subject, I think this issue can be closed.

With regards,

RKOUSTM, 2020-08-18:
ST Internal Reference: 76137

fpistm, 2020-10-09:
Hi @RKOUSTM 

In STM32duino, it is possible to redefine `VECT_TAB_OFFSET` to use a bootloader.
In a general way Bootloader is stored at the beginning of the FLASH, in that case the `VECT_TAB_OFFSET` is set to the `SCB->VTOR = FLASH_BASE | VECT_TAB_OFFSET;` where offset is set after the BL.

With the `USER_VECT_TAB_ADDRESS`  I don't understand how we can properly set it as the `VECT_TAB_OFFSET` is always 0.
This only allow to set it to SRAM or FLASH.

RKOUSTM, 2020-10-09:
Hi @fpistm,

Thank you for your contribution. The above fix is proposed to mention if we are on _``FLASH``_ or _``SRAM``_, and if we want to apply a _``different Offset value``_, as in the STM32WL, the following:
```
/* #define VECT_TAB_SRAM */
#if defined(VECT_TAB_SRAM)
#define VECT_TAB_BASE_ADDRESS   SRAM2_BASE      /*!< Vector Table base address field.
                                                     This value must be a multiple of 0x100. */
#define VECT_TAB_OFFSET         0x00008000U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x100. */
#else
#define VECT_TAB_BASE_ADDRESS   FLASH_BASE      /*!< Vector Table base address field.
                                                     This value must be a multiple of 0x100. */
#define VECT_TAB_OFFSET         0x00020000U        /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x100. */
#endif
```

Thank you again for your contribution.

With regards,

fpistm, 2020-10-09:
Thanks for the feedback @RKOUSTM 
But I think the OP request concern the `VECT_TAB_OFFSET` not if we are on `FLASH` or `SRAM`.
Anyway you answered my question and will update STM32duino to also be able to redefine  `VECT_TAB_OFFSET`

closed 2021-03-03 by RKOUSTM

theadib, 2021-03-04:
please ST 
the register holding the vector address should be written as part of the startup code in any case.
not only when USER_ADDRESS is defined.

````
SCB->VTOR = VECT_TAB_BASE_ADDRESS | VECT_TAB_OFFSET;
````