STMicroelectronics/STM32CubeF4 #32: CubeMX generated lwip.c code should support per device IP-address

theadib, 2020-07-14:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.
If one generate project with lwip middleware, it is possible to specify IP address and MAC address in the CubeMX IDE.
However noone would generate 1(one) CubeMX project per device in order to setup different IP Addresses per device.
Instead the generated code should support some kind of user customization to allow per device addresses.

**Describe the bug**
A clear and concise description of what the bug is.

For IP address the code in lwip is 
`
void MX_LWIP_Init(void)
{
 /* IP addresses initialization */
  IP_ADDRESS[0] = 192;
  IP_ADDRESS[1] = 168;
  IP_ADDRESS[2] = 123;
  IP_ADDRESS[3] = 3;
  NETMASK_ADDRESS[0] = 255;
  NETMASK_ADDRESS[1] = 255;
  NETMASK_ADDRESS[2] = 255;
  NETMASK_ADDRESS[3] = 0;
  GATEWAY_ADDRESS[0] = 0;
  GATEWAY_ADDRESS[1] = 0;
  GATEWAY_ADDRESS[2] = 0;
  GATEWAY_ADDRESS[3] = 0;
  
  /* Initilialize the LwIP stack without RTOS */
  lwip_init();

  /* IP addresses initialization without DHCP (IPv4) */
  IP4_ADDR(&ipaddr, IP_ADDRESS[0], IP_ADDRESS[1], IP_ADDRESS[2], IP_ADDRESS[3]);
  IP4_ADDR(&netmask, NETMASK_ADDRESS[0], NETMASK_ADDRESS[1] , NETMASK_ADDRESS[2], NETMASK_ADDRESS[3]);
  IP4_ADDR(&gw, GATEWAY_ADDRESS[0], GATEWAY_ADDRESS[1], GATEWAY_ADDRESS[2], GATEWAY_ADDRESS[3]);

  /* add the network interface (IPv4/IPv6) without RTOS */
  netif_add(&gnetif, &ipaddr, &netmask, &gw, NULL, &ethernetif_init, &ethernet_input);

  /* Registers the default network interface */
  netif_set_default(&gnetif);

  if (netif_is_link_up(&gnetif))
  {
    /* When the netif is fully configured this function must be called */
    netif_set_up(&gnetif);
  }
  else
  {
    /* When the netif link is down this function must be called */
    netif_set_down(&gnetif);
  }

  /* Set the link callback function, this function is called on change of link status*/
  netif_set_link_callback(&gnetif, ethernetif_update_config);

  /* Create the Ethernet link handler thread */

/* USER CODE BEGIN 3 */
   ethernetif_notify_conn_changed(&gnetif);
/* USER CODE END 3 */
}
`
As you see there thee is straight code section from the beginning until the end with no USER CODE BEGIN/END.

For MAC Address the generated code lookes like this ethernetif.c:
please note that I (the user) can customize the MAC address per device
static void low_level_init(struct netif *netif)
{ 
  HAL_StatusTypeDef hal_eth_init_status;
  
/* Init ETH */

   uint8_t MACAddr[6] ;
  heth.Instance = ETH;
  heth.Init.AutoNegotiation = ETH_AUTONEGOTIATION_ENABLE;
  heth.Init.PhyAddress = PHY_KSZ8081_PHY_ADDRESS;
  MACAddr[0] = 0x00;
  MACAddr[1] = 0x80;
  MACAddr[2] = 0xE1;
  MACAddr[3] = 0x00;
  MACAddr[4] = 0x00;
  MACAddr[5] = 0x00;
  heth.Init.MACAddr = &MACAddr[0];
  heth.Init.RxMode = ETH_RXPOLLING_MODE;
  heth.Init.ChecksumMode = ETH_CHECKSUM_BY_HARDWARE;
  heth.Init.MediaInterface = ETH_MEDIA_INTERFACE_MII;

  /* USER CODE BEGIN MACADDRESS */
   // assert the phy reset line for 5msec
    HAL_GPIO_WritePin(ETH_RST_L_GPIO_Port, ETH_RST_L_Pin, GPIO_PIN_RESET);
    HAL_Delay(5);
    HAL_GPIO_WritePin(ETH_RST_L_GPIO_Port, ETH_RST_L_Pin, GPIO_PIN_SET);
    HAL_Delay(5);
    
    memcpy(MACAddr, app_ctx.mac_address, 6);
  /* USER CODE END MACADDRESS */

  hal_eth_init_status = HAL_ETH_Init(&heth);
...
`
Currently all my customization is lost after regenerating the project.

as a solution I recommend to insert an other USER CODE section in the lwip.c like this before lwip_init():
For IP address the code in lwip is 
`
void MX_LWIP_Init(void)
{
 /* IP addresses initialization */
  IP_ADDRESS[0] = 192;
  IP_ADDRESS[1] = 168;
  IP_ADDRESS[2] = 123;
  IP_ADDRESS[3] = 3;
  NETMASK_ADDRESS[0] = 255;
  NETMASK_ADDRESS[1] = 255;
  NETMASK_ADDRESS[2] = 255;
  NETMASK_ADDRESS[3] = 0;
  GATEWAY_ADDRESS[0] = 0;
  GATEWAY_ADDRESS[1] = 0;
  GATEWAY_ADDRESS[2] = 0;
  GATEWAY_ADDRESS[3] = 0;

/* USER CODE END IPADDRESS */
/* USER CODE END IPADDRESS */

  /* Initilialize the LwIP stack without RTOS */
  lwip_init();

  /* IP addresses initialization without DHCP (IPv4) */
  IP4_ADDR(&ipaddr, IP_ADDRESS[0], IP_ADDRESS[1], IP_ADDRESS[2], IP_ADDRESS[3]);
  IP4_ADDR(&netmask, NETMASK_ADDRESS[0], NETMASK_ADDRESS[1] , NETMASK_ADDRESS[2], NETMASK_ADDRESS[3]);
  IP4_ADDR(&gw, GATEWAY_ADDRESS[0], GATEWAY_ADDRESS[1], GATEWAY_ADDRESS[2], GATEWAY_ADDRESS[3]);

  /* add the network interface (IPv4/IPv6) without RTOS */
  netif_add(&gnetif, &ipaddr, &netmask, &gw, NULL, &ethernetif_init, &ethernet_input);
...
`

This way also IP Address is customizable by user.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RKOUSTM, 2020-07-16:
Hi @theadib,

Thank you for your contribution. According to our analysis, this issue is rather related to CubeMX, may I ask you to submit it to the ST Community in the dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx)? There, you will hopefully find people from our CubeMX development teams who will take in charge your request.

Thank you again for your contribution.

With regards,

closed 2020-07-16 by RKOUSTM

theadib, 2020-07-18:
Dear RKOUSTM,

thanks for your response and guidance.

Do you know if or when ST will create a project for STMCubeMX on github as
well to track issues.

Regards, Adib.

RKOUSTM, 2020-07-24:
Hi @theadib,

First of all we would like to thank you for your interest.  However, aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the Cube [MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section.

Thank you for your comprehension.

With regards,