STMicroelectronics/STM32CubeF4 #34: RTOS + USB MSC. Message Queue overflow

DanielKirsch, 2020-07-21:
**Describe the set-up**
 * STM32F411 Discovery kit
 * FreeRTOS + USB Host + MSC Class + FatFs

**Describe the bug**
The USB driver is putting a way to much in the message queue of the OS. The OS is returning an error code that the memory can't be reserved anymore. All error codes are been ignored from the OS, that's why it is still "working". 

**How To Reproduce**
1. Activate USBH_USE_OS 
2. Set in usbh_def.h MSGQUEUE_OBJECTS to an high value. 250 for example. 
3. You should see that after a while all 250 entries are been used, when an USB is connected. 
3. Another way is to add in cmsis_os.c in the function of osMessagePut(...) an error reaction if xQueueSend wasn't successful. 

**Additional context**
The USBH_Process shall only add something in the queue if really needed and if not already done.

DanielKirsch, 2020-07-22:
I think a good solution would anyway be to replace all 

> #if (USBH_USE_OS == 1U)
        phost->os_msg = (uint32_t)USBH_PORT_EVENT;
#if (osCMSIS < 0x20000U)
        (void)osMessagePut(phost->os_event, phost->os_msg, 0U);
#else
        (void)osMessageQueuePut(phost->os_event, &phost->os_msg, 0U, NULL);
#endif
#endif

with a single define like USBH_OS_TASKTRIGGER ( USBH_PORT_EVENT ) that can then be configured and the user has the option to replace it with an more efficient function then osMessageQueue.

ALABSTM, 2020-07-22:
Hi @DanielKirsch,

Thank you for this report and thank you also for the solution suggested. You request has been forwarded to our development teams. I will get back to you as soon as they provide me with their feedback.

With regards,

ALABSTM, 2021-01-18:
Hi @DanielKirsch,

First of all we would like to apologize for the time it takes to answer sometimes. Please allow me to introduce @CHAMSTM who will look deeper into what you reported.

Thank you for your patience and again thank you for your contribution.

With regards,

ALABSTM, 2021-05-27:
Hi @DanielKirsch,

Our development teams confirmed your point. They will deeper analyze your proposal and share a fix in a future release. Thank you again for your contribution.

With regards,

ALABSTM, 2021-05-27:
ST Internal Reference: 107774

GiulioDallaVecchia, 2021-10-12:
Hi ST Team,

any news about that?