STMicroelectronics/STM32CubeF4 #35: HAL_MMC_InitCard fails since Release v1.24.2

GregorBader, 2020-07-23:
Beginning with Release v1.24.2 of STM32CubeF4 the initialization of MMC in function `HAL_MMC_InitCard()` fails.

The reason for the failure is, that the required power up waiting time is moved from `HAL_MMC_InitCard()` into `SDIO_PowerState_ON()` but the MMC clock enable is still after `SDIO_PowerState_ON()` resulting in no MMC clock during the waiting time!

[image]

The disable/enable of the MMC clock should be reworked.

ALABSTM, 2021-03-08:
Hi @GregorBader,

It seems we missed this issue you have sent since a while. Our apology for the inconvenience. We will forward it to our development teams for analysis. We will get back to you as soon as we have an answer.

With regards,

ALABSTM, 2021-08-02:
Hi @Microtronics-DC-FW,

Your point has been confirmed by our development team. It will be logged into our internal database and a fix will be published in a future release. Thank you for your report and excuse this somewhat late reply.

With regards,

ALABSTM, 2021-08-02:
ST Internal Reference: 111415

RKOUSTM, 2022-03-11:
Hi @Microtronics-DC-FW,

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-03-11 by RKOUSTM