STMicroelectronics/STM32CubeF4 #37: does not stop USB device correctly in HAL_PCD_Stop()

theadib, 2020-09-03:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.
I use STM32F437VGT powered by battery (not USB).
I use CubeMX for setup and Initialisation
USB_FS is  is configured as CDC device.
MX_USB_DEVICE_Init is called by application not by main().

**Describe the bug**
A clear and concise description of what the bug is.
After detecting the USB cable is detached the application calls usbd_core.c:USBD_DeInit() function.
The goal is to detach USB from the USB pins to avoid unneeded PULL UP voltage on that pins.

The Problem is that the function USBD_DeInit() aborts with errorr code halfway and does not release USB from USB Pins.
1) USBD_LL_Stop does report USBD_TIMEOUT instead USBD_OK
2) function return 
3)  USBD_LL_DeInit is not called
   that would lead to USB detach USB pins

**How To Reproduce**
1. Indicate the global behavior of your application project.
1) start main
2) attach USB cable
3) call MX_USB_DEVICE_Init() -> PC does enumeration
4) detach USB cable
5) call USBD_DeInit -> function returns halfway and does not call USBD_LL_DeInit

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
**
1) in file stm32f4xx_ll_usb.c line:1070 function USB_StopDevice is wrongly implemented.
2) in file stm32f4xx_ll_usb.c line:477 function USB_FlushRxFifo is wrongly implemented.
3) in file stm32f4xx_ll_usb.c line:454 function USB_FlushTxFifo is wrongly implemented.
4) the silicon has defects
**

1)  does not respect the real number of endpoints in the specific device
```
HAL_StatusTypeDef USB_StopDevice(USB_OTG_GlobalTypeDef *USBx)
...
  /* Clear Pending interrupt */
  for (i = 0U; i < 15U; i++)
  {
    USBx_INEP(i)->DIEPINT = 0xFB7FU;
    USBx_OUTEP(i)->DOEPINT = 0xFB7FU;
  }
...
```
according to datasheet for STM32F437 on the FS device i is in range 0..3 and for HS device i in range 0..7
So there must be differentiation of FS and HS device and respect the range.

1) should ignore errors on stopping
```
HAL_StatusTypeDef USB_StopDevice(USB_OTG_GlobalTypeDef *USBx)
...
  /* Flush the FIFO */
  ret = USB_FlushRxFifo(USBx);
  if (ret != HAL_OK)
  {
    return ret;
  }

  ret = USB_FlushTxFifo(USBx,  0x10U);
  if (ret != HAL_OK)
  {
    return ret;
  }
```
The initial problem could be avoided if there is no test of return value on USB_FlushRxFifo(USBx) and USB_FlushTxFifo(USBx,  0x10U).
So please do not test the return in the StopDevice() function.

2)  the function deserves a synchronisation barrier and more testing
```
HAL_StatusTypeDef USB_FlushRxFifo(USB_OTG_GlobalTypeDef *USBx)
{
...
    }
    __DSB();
  }
  while ((USBx->GRSTCTL & USB_OTG_GRSTCTL_RXFFLSH) == USB_OTG_GRSTCTL_RXFFLSH);

  return HAL_OK;
}
```
also in the registerdescription is is pointed out that care must be taken when accessing the flush function is the fifo is still in use.
I do not see such test.

3) the function deserves a synchronisation barrier and more testing 
```
HAL_StatusTypeDef USB_FlushTxFifo(USB_OTG_GlobalTypeDef *USBx, uint32_t num)
...
    }
    __DSB();
  }
  while ((USBx->GRSTCTL & USB_OTG_GRSTCTL_TXFFLSH) == USB_OTG_GRSTCTL_TXFFLSH);

```
also in the registerdescription is is pointed out that care must be taken when accessing the flush function is the fifo is still in use.
I do not see such test.

4)  silicon defect 
the problem arises when flushing the RXFIFO.
The function times-out and return error.
`
USBx->GRSTCTL = USB_OTG_GRSTCTL_RXFFLSH; 
`
after setting the bit, the hardware does not return the register bit.

According to datasheet after setting the RXFIFO flush bit.
_"""
The application must wait until the bit is cleared before performing any other operations. This
bit requires 8 clocks (slowest of PHY or AHB clock) to clear.
"""_
The bit should be safely cleared within 20000 queries that is implemented.

Hope this helps to improve the driver, regards Adib.

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2020-09-16:
Hi @theadib,

Thank you for all these details and proposals. You report will be forwarded to our development teams for deeper analysis. I will get back to you as soon as I have their feedback.

With regards,

ALABSTM, 2021-05-25:
Hi @theadib,

I hope you are fine. Our development teams confirmed the points you raised. Fixes will be made available in a future release. Thank you again for having reported.

With regards,

ALABSTM, 2021-05-25:
ST Internal Reference: 107637

ALABSTM, 2021-10-20:
Hi @theadib,

Back to you with some updates. Our development teams confirmed almost all the points you reported:
* In the function `USB_HostInit()` and `USB_StopHost()`, the values returned by the FIFO flush functions are not checked.
* In the @brief section of the `USB_FlushTxFifo()`, the function name is incorrect.
* According to REFMAN RM0390, bits `TXFFLSH` and `RXFFLSH`, respectively, must not be set while the core is writing to or reading from the Tx and Rx FIFOs, respectively. Such check is not done.

All these points have been dealt with and fixes will be made available in a future release.

Only one point will be kept unchanged.
* In the function `USB_StopDevice()`, the max number of endpoints is hard-coded (`15U`) while, according to the REFMAN, it depends on the device's part-number.

https://github.com/STMicroelectronics/STM32CubeF4/blob/4aba24d78fef03d797a82b258f37dbc84728bbb5/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c#L1116-L1121

This last point has been confirmed **not** to be an issue and we prefer to keep that approach to ensure all IN and OUT EPs' registers are cleared.

Thank you again for this detailed report.

With regards,

ALABSTM, 2022-04-19:
Hi @theadib,

Hope you're fine. Just to inform you the points reported here have been fixed in the frame of version **1.27.0** published a couple of weeks ago.

Thank you again for your contribution.

With regards,

closed 2022-04-19 by ALABSTM