STMicroelectronics/STM32CubeF4 #41: USB Host implementation broken when sending transfers that are larger than max packet size

eranrund, 2020-09-15:
I ran into the following issues when trying to send a transaction that is split into multiple transfers.

1) The code [here](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c#L956-L974) does not take into account the FIFO size.

2) For large transfers, the code [here](https://github.com/STMicroelectronics/STM32CubeF4/blob/9f5ea7421d22c724516d00218b350d1e09a54a2c/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c#L1819) enables the ptxfempty interrupt, but the interrupt handles ignores it as seen in https://github.com/STMicroelectronics/STM32CubeF4/blob/9f5ea7421d22c724516d00218b350d1e09a54a2c/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_hcd.c#L521-L524

Am I missing something or this functionality just super broken?

ALABSTM, 2020-09-16:
Hi @eranrund,

Your question will be forwarded to our development teams. I will get back to you as soon as they provide me with their answer.

With regards,

eranrund, 2020-09-16:
Thank you @ALABSTM

ALABSTM, 2022-04-25:
ST Internal Reference: 126900

ALABSTM, 2022-04-25:
Hi @eranrund,

I hope you are doing well. Back to you about this point you raised. According to our development team:
* Currently, transfers larger than the maximum packet size are handled at MW library level. This means the user has to use whether the ST USB lib or the Azure USBX lib.
* The ST USB HAL driver does not handle this case for the moment.
* Hence, in case another lib is used along with the ST USB HAL driver, such transfers are very likely not to be handled.

A request has been submitted to have this updated. The development teams will deeper analyze it before deciding whether to go for such change or not. I will keep you informed.

With regards,

ALABSTM, 2023-07-26:
Hi @eranrund,

Back to you about this point you raised. Unfortunately, our development teams do not plan to implement such enhancement. Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2023-07-26 by ALABSTM