STMicroelectronics/STM32CubeF4 #42: STM32F412Cx USART3 contradiction between CMSIS and HAL

boraozgen, 2020-09-18:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
Custom board with STM32F412CG.
Mbed OS 6.2.0 with GCC ARM 9-2020-q2-update

**Describe the bug**
Cannot link `serial_api`, since there is a contradiction between CMSIS and HAL:

In CMSIS `stm32f412cx.h` `USART3_BASE` is defined. 

In RCC HAL, USART3 functions are not defined for `STM32F412Cx`.

https://github.com/STMicroelectronics/STM32CubeF4/blob/9f5ea7421d22c724516d00218b350d1e09a54a2c/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_rcc_ex.h#L5448-L5449

This causes linker issues for example when trying to link `serial_api.o` of the STM HAL in Mbed OS due to these lines:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/serial_api.c#L246-L252

I believe this is not an Mbed issue but an HAL issue. 

**How To Reproduce**
Try to build an Mbed OS 6.2.0 project for STM32F412CG.

**Additional context**
It is fixed when `STM32F412Cx` is added to the conditions for all USART3 functions.

**Screenshots**
N.a.

fpistm, 2020-09-18:
Hi @boraozgen 

I've already raised this issue internally as I've met this issue with STM32duino. I've made a patch while waiting for it to be fixed
in the official release.
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/CI/utils/patch/HAL/F4/0001-F4-Fix-HAL_RCC_USART3-definition-for-STM32F412Cx.patch

ST Internal Reference: 82376

boraozgen, 2020-09-18:
Great, thank you for the information. Looking forward to the next release.

jeromecoutant, 2020-09-18:
@boraozgen 
You could propose a patch in mbed-os repo, I will review and it could be merged there.

fpistm, 2020-09-18:
Just FYI, issue comes from the fact STM32F412Cx define is to manage the STM32F412 UFQFPN48 part numbers, for these devices it have a hardware limitation (Note 2 in datasheet page 15):

_Limited application for the USART3 since RX is not available for the UFQFPN48._

So there is only the USART3 TX available which van be used for half duplex.

fpistm, 2020-09-18:
@boraozgen 

> Looking forward to the next release.

About this I cannot guarantee that a fix will be in the next release. I'm only a user of the HAL and CMSIS for the STM32duino GitHub organization 😉 I would just inform that this issue was already tracked internally and the fix I'm using temporarily.

I leave @ALABSTM or @RKOUSTM comment on this.

ASELSTM, 2020-09-21:
Hi @boraozgen,

Thank you for your contribution. As @fpistm said the issue has been tracked and fixed internally. The fix will be made available in the frame of a future release.  Unfortunately, we cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

Thank you also @fpistm, for your contribution and for your reactivity as well as for the given clarifications.

With regards,

ASELSTM, 2020-09-21:
ST Internal Reference: 82376

ALABSTM, 2020-09-21:
Reopened till the fix is released on GitHub.

ASELSTM, 2021-03-03:
Hi @boraozgen,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.26.0 of the STM32CubeF4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM