STMicroelectronics/STM32CubeF4 #45: STM32 F411E I2C hang

WojciechBernat, 2020-09-25:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
STM32F411E-DISCO
STM32CubeIDE Version: 1.2.1 Build: 5190_20200115_1224 (UTC)

**Describe the bug**
When I was using HAL_I2C_Mem_Read when I detected problem with I2C_RequestMemoryRead. 
I2C_RequestMemoryRead always returned HAL_ERROR. 
When I was creating new project I changed frimware version for F411 from 1.24.2 to 1.24.0.
This action fixed this issue. 

**How To Reproduce**
1. Indicate the global behavior of your application project.
I have no I2C communiction. 

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
stm32f4xx_hali2c.h
stm32f4xx_hali2c.c

3. The use case that generates the problem.
Communication by i2c with LSM303DLHC on discovery board.
Use HAL_I2C_Mem_Read -  this function always returns HAL_ERROR

4. How we can reproduce the problem.
Generate project in CubeIDE for LSM303 communication. Use HAL_I2C_Mem_Read form 1.24.2 firmware version.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ASELSTM, 2020-10-06:
Hi @WojciechBernat, 

Thank you for your contribution.

Could you try with the latest firmware release **(V1.25.1)** and check if you still have the same problem ? 

With regards,

WojciechBernat, 2020-10-11:
Hi

Thank you for new software.
I downloaded software and replaced HAL's spi driver.
I ran tests and it has worked fine.

With regards,
Wojciech Bernat

WojciechBernat, 2020-10-11:
Hi

I made a mistake. I'm working on few projects in the same time and mistake
project.
I check how new I2C driver is working with my project. It's working
correct.
I'm very sorry for mistake.

With regards,
Wojciech Bernat

niedz., 11 paź 2020 o 10:50 Wojciech Bernat <wojciechbernat95@gmail.com>
napisał(a):

ASELSTM, 2020-10-12:
Hi @WojciechBernat, 

Glad to know that everything is working well.  Please allow me then to close this thread. Thank you again for your contribution.

With regards,

closed 2020-10-12 by ASELSTM