STMicroelectronics/STM32CubeF4 #48: Problem in using SMBus driver

karnpinyo, 2020-11-05:
Hi,

I have some issues using SMBus. The source file which I suspect to be the cause of problems is _stm32f4xx_hal_smbus.c_. Use cases which I found the issues are described below.
1. If I use `HAL_SMBUS_Master_Transmit_IT` with `SMBUS_FIRST_AND_LAST_FRAME_NO_PEC` XferOptions, STOP is not generated. I suspect the condition in line 1934 and 2020. Shouldn't all conditions be ANDed together?
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c#L1933-L1936
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c#L2019-L2021

2. If I use `HAL_SMBUS_Master_Receive_IT` with `SMBUS_LAST_FRAME_NO_PEC` XferOptions, STOP is also not generated. I suspect the condition in line 2169. Shouldn't `SMBUS_LAST_FRAME_NO_PEC` be deleted from the condition?
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c#L2168-L2170

3. Not a bug, but I found that `HAL_SMBUS_ErrorCallback` is called twice in line 2714.
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c#L2713-L2719

Regards

ASELSTM, 2020-12-07:
Hi @karnpinyo,

Thank you for this report. Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-12-08:
Hi @karnpinyo,

Our development teams confirmed the issue you pointed out :

1. The check condition is indeed always correct and the first statement is always executed so that stop is never generated. The condition should be corrected. However, the fix proposed will lead to the same behavior, the condition should be corrected differently. Our development team is working on this and will propose a correction. 

2.  In fact, `SMBUS_LAST_FRAME_NO_PEC` should be deleted from the condition which will be : 
```diff
-if ((CurrentXferOptions == SMBUS_NEXT_FRAME) || (CurrentXferOptions == SMBUS_FIRST_FRAME) || (CurrentXferOptions == SMBUS_LAST_FRAME_NO_PEC)) 
+if ((CurrentXferOptions == SMBUS_NEXT_FRAME) || (CurrentXferOptions == SMBUS_FIRST_FRAME)) 
```

3. The first callback call will be deleted to avoid calling `HAL_SMBUS_ErrorCallback` twice :

```diff
-HAL_SMBUS_ErrorCallback(hsmbus); 
 #if (USE_HAL_SMBUS_REGISTER_CALLBACKS == 1) 
     hsmbus->ErrorCallback(hsmbus); 
 #else 
     HAL_SMBUS_ErrorCallback(hsmbus); 
 #endif /* USE_HAL_SMBUS_REGISTER_CALLBACKS */ 
```

The fixes will be implemented and made available in a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-12-08:
ST Internal Reference: 97820

ASELSTM, 2020-12-08:
ST Internal Reference: 97838

ASELSTM, 2021-03-03:
Hi @karnpinyo,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.26.0 of the STM32CubeF4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM