STMicroelectronics/STM32CubeF4 #51: RAM optimization for USB CDC

jrahlf, 2020-12-06:
The following declaration wastes 448 Byte of RAM on all devices where CDC is used with a USB FS spec.
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc/usbd_cdc.h#L113

The documentation suggests you could modify ```CDC_DATA_HS_MAX_PACKET_SIZE``` but that would only be a workaround.

Possible fix:
```c
typedef struct
{
#ifdef USBD_NO_HS          /* can be set by cubeMX or provided by user as project configuration */
   uint32_t data[CDC_DATA_FS_MAX_PACKET_SIZE / 4U];      /* Force 32bits alignment */
#else
   uint32_t data[CDC_DATA_HS_MAX_PACKET_SIZE / 4U];      /* Force 32bits alignment */
#endif
```

ALABSTM, 2021-01-20:
Hi @jrahlf,

Thank you for your proposal. It will be logged into our internal database so our development teams consider it in the frame of a future version of the library.

With regards,

ALABSTM, 2021-01-20:
ST Internal Reference: 99825

jrahlf, 2021-03-14:
Hi @ALABSTM
can you tell where `USE_USB_FS` is supposed to be defined?
I created a STM32F4 project which uses USB FS with CDC and the #define does not exist. It is however used in `usbd_core.c`. In the debugger I see that the HS descriptor is used, even though the USB is FS only.

```c
#ifdef USE_USB_FS
  pdev->pConfDesc = (void *)pdev->pClass->GetFSConfigDescriptor(&len);
#else /* USE_USB_HS */
  pdev->pConfDesc = (void *)pdev->pClass->GetHSConfigDescriptor(&len);
#endif /* USE_USB_FS */
```

This seems like a bug to me?
It still works nicely so nobody has noticed it before. Seems like the USB driver or the computer driver somehow fixes this, otherwise the mismatch in packet length (512 vs 64) would cause problems.

ALABSTM, 2021-03-23:
Hi @jrahlf,

Both `USE_USB_HS` and `USE_USB_FS` are supposed to by defined by the user at command line level or at project's options-level using your preferred toolchain.

This is already the case with the provided USBD applications like the [CDC_Standalone](https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32446E_EVAL/Applications/USB_Device/CDC_Standalone) for instance. Below are the lines from the different project files where the symbol is defined.

https://github.com/STMicroelectronics/STM32CubeF4/blob/d599a824dfe257bee2aeff403caf592da816b126/Projects/STM32446E_EVAL/Applications/USB_Device/CDC_Standalone/EWARM/USBD_CDC.ewp#L191

https://github.com/STMicroelectronics/STM32CubeF4/blob/d599a824dfe257bee2aeff403caf592da816b126/Projects/STM32446E_EVAL/Applications/USB_Device/CDC_Standalone/MDK-ARM/Project.uvprojx#L366

https://github.com/STMicroelectronics/STM32CubeF4/blob/d599a824dfe257bee2aeff403caf592da816b126/Projects/STM32446E_EVAL/Applications/USB_Device/CDC_Standalone/SW4STM32/STM32446E-EVAL_USBD-HS/.cproject#L32

https://github.com/STMicroelectronics/STM32CubeF4/blob/d599a824dfe257bee2aeff403caf592da816b126/Projects/STM32446E_EVAL/Applications/USB_Device/CDC_Standalone/SW4STM32/STM32446E-EVAL_USBD-FS/.cproject#L32

With regards,

jrahlf, 2021-03-25:
Hi @ALABSTM,
thanks for your feedback.
It seems that the value is not defined when a new project is generated via CubeIDE. I opened a thread in the forums about this issue: https://community.st.com/s/question/0D53W00000ewIM2SAM/missing-define-useusbfs-useusbhs-for-usb-device-library

Interestingly, the latest USB library reverses the logic, so that by default it uses USB FS if the neither value is undefined.