STMicroelectronics/STM32CubeF4 #52: RAM optimization for USB CDC (2)

jrahlf, 2020-12-06:
1. Why are these not marked as const:
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L255
Because of non-constness, they are placed in RAM. However, they are not modified at runtime so making them const would save 3*67Bytes in RAM.

And while we are looking at this file:

2. Why is there even a descriptor for "other speed" ? It sets the bDescriptorType field to 7, I cannot find any documentation for this value on the internet. Why is it not 0x2? 

3. The documentation for the different fields in the descriptors is different for "high speed" / "full speed" / "other speed". Please consolidate that.

ALABSTM, 2021-01-20:
Related to issue [STM32CubeF4#10](https://github.com/STMicroelectronics/STM32CubeF4/issues/10).

ALABSTM, 2021-01-20:
Hi @jrahlf,

Thank you for this other optimization proposal. It will be logged into our internal database too. Identified descriptors that could be placed in read-only region are listed below.

With regards,

https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L116
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L159
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L255
https://github.com/STMicroelectronics/STM32CubeF4/blob/2d5b78e2da98fb44e028d87f4ff815427991affc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L349

ALABSTM, 2021-01-20:
ST Internal Reference: 99842

ALABSTM, 2021-01-20:
Hi @jrahlf,

Regarding this question below, may I ask you to clarify which "documentation" you are referring to?

> 3. The documentation for the different fields in the descriptors is different for "high speed" / "full speed" / "other speed". Please consolidate that.

Thank you,

jrahlf, 2021-01-21:
Hi @ALABSTM , please see the attached image regarding the differences in the documentation. I think only the "OtherSpeed" descriptor is affected. E.g. one documentation says 100mA and the other one says 0mA even though the value is the same. There also other subtle differences, .e.g differences in whitespace or 
`/* bInterval: ignore for Bulk transfer */` vs 
`/* bInterval */`.

As I now understand, the OtherSpeed desciptor is to inform the host about the possible Highspeed configuration, if the device is running in FullSpeed mode. If that is the case, the **wMaxPacketSize** field should be set to 512 instead of 64 (for the OtherSpeed descriptor). Can you check this?

[image]

ALABSTM, 2021-01-22:
ST Internal Reference: 99959

ALABSTM, 2021-01-22:
Hi @jrahlf,

Thank you for the clarification. The point has been logged into our internal database. Regarding your other question below, I let @CHAMSTM comment.

> 2. Why is there even a descriptor for "other speed" ? It sets the bDescriptorType field to 7, I cannot find any documentation for this value on the internet. Why is it not 0x2?

With regards,

ALABSTM, 2021-02-23:
Hi @jrahlf,

I hope you are doing well. According to our development teams, a couple of use-cases have been identified where the user could modify some fields of the device configuration descriptor at run-time preventing from moving its location into read-only memory.

With regards,

ALABSTM, 2021-06-21:
Hi @jrahlf,

I hope you are doing well.
* Regarding moving the device configuration descriptor to a "read-only" memory region, this is not possible according to our development teams. Indeed, the descriptor is stored in `pdev->pConfDesc` and, depending on the use-case, some fields could be changed at run time. With a simple software disconnect-then-connect operation, the new descriptor will be shared to the host.
* Regarding the documentation found in the comments, it has been updated in the frame of **v2.8.0** of the USB Device Library found in the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository.

Thank you again for having shared this proposal. Please allow me to close this issue now. Thank you for your comprehension.

Regards,
Aymen

closed 2021-06-21 by ALABSTM