STMicroelectronics/STM32CubeF4 #54: Buffer overrun in USB_WritePacket if length not multiple of 4

jrahlf, 2020-12-28:
If the input length is not a multiple of 4, the function overruns the buffer by 3, 2 or 1 byte.
This could trigger an exception (HardFault/BusFault), e.g. if one passes a buffer pointing to the last byte in RAM and length=1.

https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c#L965

Other devices are possible affected as well.

ALABSTM, 2021-04-21:
Hi @jrahlf,

Thank you for having reported. Our development teams said the case you described is relevant, although very odd. They also said such an aspect is not intended to be handled at driver's level, rather at user application one. Hence, it is up to the user when designing their applications to ensure such a case does not happen.

Thank you again for your report and thank you for your comprehension.

With regards,

ALABSTM, 2021-06-30:
Hi @jrahlf,

Please allow me to close this issue. Thank you again for your comprehension.

With regards,

closed 2021-06-30 by ALABSTM

bmcdonnell-fb, 2021-06-30:
@ALABSTM it would be better to document the behavior in the function-level comments, that the function rounds up if necessary to the next multiple of 4 bytes.