STMicroelectronics/STM32CubeF4 #55: Move GPIO read/write/toggle into implementation into header

jrahlf, 2021-01-10:
I suggest to move the implementation of GPIO read/write/toggle into the header, so that the compiler (I assume GCC here) can optimize the code better without LTO. Without LTO, GCC will always insert a function call, which makes bit-banging or fast GPIO polling slower than necessary. For fast bit-banging or fast GPIO polling the overhead from the function calls actually create a measurable difference in speed.

https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c#L377

Side note regarding LTO:

The current CubeIDE ships with a bug, so that LTO can only be used in conjunction with a workaround, I already posted about that here: https://community.st.com/s/question/0D53W00000MjO4DSAV/workaround-for-gcc-lto-bug
Unfortunately, there was no respone from ST.

jrahlf, 2021-01-11:
They are also prime candidates for #10

ASELSTM, 2021-01-15:
Hi @jrahlf,

Thank you for this report.

We regret to inform you that your request can't be accepted.  We can't move the implementation of these functions to the header file in order to maintain the driver compatibility with the rest of STM32 platforms. Moreover, the deployment on all series is not conceivable,  the impact of this change is very important. Therefore, we regret to decline your request. Thank you for your understanding.

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2021-01-15 by ASELSTM