STMicroelectronics/STM32CubeF4 #58: HAL RTC Output

KKobel, 2021-01-22:
It appears that if the output bit is set (RTC_CR_COE) it doesn't get cleared when the RTC is initialized.

In the library code:

HAL_StatusTypeDef HAL_RTC_Init(RTC_HandleTypeDef *hrtc)
{

...

 else
 {
   /* Clear RTC_CR FMT, OSEL and POL Bits */
   hrtc->Instance->CR &= ((uint32_t)~(RTC_CR_FMT | RTC_CR_OSEL | RTC_CR_POL));
   /* Set RTC_CR register */
   hrtc->Instance->CR |= (uint32_t)(hrtc->Init.HourFormat | hrtc->Init.OutPut | hrtc->Init.OutPutPolarity);
 
   /* Configure the RTC PRER */
   hrtc->Instance->PRER = (uint32_t)(hrtc->Init.SynchPrediv);
   hrtc->Instance->PRER |= (uint32_t)(hrtc->Init.AsynchPrediv << 16U);
 
   /* Exit Initialization mode */
   hrtc->Instance->ISR &= (uint32_t)~RTC_ISR_INIT;
 
...
 
}

Should the above line be:

   hrtc->Instance->CR &= ((uint32_t)~(RTC_CR_FMT | RTC_CR_OSEL | RTC_CR_POL | RTC_CR_COE));
 
Karl

ALABSTM, 2021-02-10:
Hi @KKobel,

Thank you for your question. First of all, please note that the calibration output feature is handled using the two functions below:
https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc_ex.c#L1469
https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc_ex.c#L1508

Regarding bit `RTC_CR_COE` now, it is cleared at reset, as indicated in the reference manual **RM0090**. Hence, resetting it through the `HAL_RTC_Init()` function is not necessary.

[image]

In case it has been previously set and you need to reset it:
* Whether you call the `HAL_RTC_DeInit()` function which will clear the whole `RTC_CR register`
https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc.c#L411
* Or you simply use the dedicated function `HAL_RTCEx_DeactivateCalibrationOutPut()` above mentioned.

I hope this answers your question. Do not hesitate otherwise.

With regards,

ALABSTM, 2021-02-22:
Hi @KKobel,

I hope you are fine. Any feedback after my previous comment? 

Thanks,

ALABSTM, 2021-03-08:
Hi @KKobel,

I hope your are doing well. I hope my comment above answers your question. Please allow me to close this issue now. You can reopen it in the contrary case.

With regards,

closed 2021-03-08 by ALABSTM

KKobel, 2021-03-08:
Yes please.
I have no issue with it as I didn't intentionally set RTC Output to the 
enabled state.
I was just pointing out that you could set the bit in the init, but not 
clear it.

Thanks,
Karl