STMicroelectronics/STM32CubeF4 #59: RAM optimization for USB CDC (3)

theadib, 2021-02-07:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
USB_CDC receive buffers are declared in the STM_CubeMX.
Those Buffers are not really used in User application.

**Describe the bug**
A clear and concise description of what the bug is.
STM_CubeMX generates **usbd_cdc_if.c**
There the buffers

```
/* USER CODE BEGIN PRIVATE_DEFINES */
/* Define size for the receive and transmit buffer over CDC */
/* It's up to user to redefine and/or remove those define */
#define APP_RX_DATA_SIZE  1024
#define APP_TX_DATA_SIZE  1024
/* USER CODE END PRIVATE_DEFINES */
```
...
```
/* Create buffer for reception and transmission           */
/* It's up to user to redefine and/or remove those define */
/** Received data over USB are stored in this buffer      */
uint8_t UserRxBufferFS[APP_RX_DATA_SIZE];

/** Data to send over USB CDC are stored in this buffer   */
uint8_t UserTxBufferFS[APP_TX_DATA_SIZE];

```
first and only use in the Init Function
```
/**
  * @brief  Initializes the CDC media low layer over the FS USB IP
  * @retval USBD_OK if all operations are OK else USBD_FAIL
  */
static int8_t CDC_Init_FS(void)
{
  /* USER CODE BEGIN 3 */
  /* Set Application Buffers */
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserTxBufferFS, 0);
  USBD_CDC_SetRxBuffer(&hUsbDeviceFS, UserRxBufferFS);
  return (USBD_OK);
  /* USER CODE END 3 */
}
```
TxBuffer: To give BufferAddress when size is Zero is useless.
RxBuffer: the buffer is only handled when following the call to USBD_CDC_ReceivePacket()
     so this call to SetRxBuffer is completely useless.

further calls to Transmit use the user buffer not the buffer from the CDC Interface file
```
no further use of TX_Buffer on sending
uint8_t CDC_Transmit_FS(uint8_t* Buf, uint16_t Len)
{
  uint8_t result = USBD_OK;
  /* USER CODE BEGIN 7 */
  USBD_CDC_HandleTypeDef *hcdc = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
  if (hcdc->TxState != 0){
    return USBD_BUSY;
  }
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
  result = USBD_CDC_TransmitPacket(&hUsbDeviceFS);
  /* USER CODE END 7 */
  return result;
}

```
further calls to Receive use the user buffer not the buffer from the CDC Interface file
```
no further use of RX_Buffer on sending
static int8_t CDC_Receive_FS(uint8_t* Buf, uint32_t *Len)
{
  /* USER CODE BEGIN 6 */
  USBD_CDC_SetRxBuffer(&hUsbDeviceFS, &Buf[0]);
  USBD_CDC_ReceivePacket(&hUsbDeviceFS);
	return (USBD_OK);
  /* USER CODE END 6 */
}
```
**THIS IS MY PROPOSAL**

All in All:
- please remove the UserRxBufferFS and UserTxBufferFS completely.
- modify the CDC_Init_FS() function as below,
- this will free as much RAM as possible.
- please note that there are requirements on that UserRxBufferFS[
  It seems the Size of the buffer is predefined and 
```
new Init function:
static int8_t CDC_Init_FS(void)
{
  /* USER CODE BEGIN 3 */
  /* Set Application Buffers */
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, NULL, 0);
  USBD_CDC_SetRxBuffer(&hUsbDeviceFS, NULL);
  return (USBD_OK);
  /* USER CODE END 3 */
}
```
please note: WARNING
It is strange that Buffersize is not part of the API in USBD_CDC_SetRxBuffer() where it is in USBD_CDC_SetTxBuffer().
looking into the code in file usbd_cdc.c assumes that buffersize is as much as CDC_DATA_HS_OUT_PACKET_SIZE / CDC_DATA_FS_OUT_PACKET_SIZE.
```
USBD_LL_PrepareReceive(pdev, CDC_OUT_EP, hcdc->RxBuffer,
                                 CDC_DATA_HS_OUT_PACKET_SIZE);
```
**THIS IS MY PROPOSAL**

1. change the API to include buffer size.
2. create a static buffer for receive only

[image]

jrahlf, 2021-02-11:
In my opinion ST should provide a proper buffer management for input & output. 
The standard case is to generate data in the user application which should then transparently get sent out whenever the USB becomes free.

See https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/usb/cdc/usbd_cdc_if.c as an example.

theadib, 2021-02-12:
I agree to jrahlf, good design and usefull API are key points. (const correct transmit buffer ;-) )
I hope that ST got the message that the buffer size declaration in CubeMX is useless here.

Adib.

jrahlf, 2021-03-04:
HI @theadib,
I created a modified version of the CDC here: https://github.com/jrahlf/stm32_usb_cdc_improved_if
It is inspired by the Arduino stm32 project, some things are still WIP. Maybe you want to have a look.

MichaelSweden, 2021-04-25:
I have get rid of UserTxBufferFS in my application and don't want any extra buffer of transmission (which I guess was the intention of this buffer, i.e. an intermediate storage). Please note that in:
```
uint8_t CDC_Transmit_FS(uint8_t* Buf, uint16_t Len)
{
  uint8_t result = USBD_OK;
  /* USER CODE BEGIN 7 */
  USBD_CDC_HandleTypeDef *hcdc = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
  if (hcdc->TxState != 0){
    return USBD_BUSY;
  }
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
  result = USBD_CDC_TransmitPacket(&hUsbDeviceFS);
  /* USER CODE END 7 */
  return result;
}
```

The content of the user buffer (Buf), needs to be valid until USB driver has sent the data. However there is no function to check or wait for this. I did like this in my application:
```
static uint8_t CdcTransmitOngoing()
{
    USBD_CDC_HandleTypeDef *hCdc = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
    return (hCdc->TxState != 0) ? 1 : 0;
}

static void CdcTransmit(uint8_t* Buf, uint16_t Len)
{
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
    USBD_CDC_TransmitPacket(&hUsbDeviceFS);
    while (CdcTransmitOngoing())
        ; // Need to wait until the buffer is read out
}
```
PS. _The standard case is to enqueue data from user application side which then transparently gets sent out whenever the USB becomes free._ Might not be a good idea in this type of devices, since it require extra memory and copy of data. In my use case I actually will use nearly all RAM in the device for a big buffer that shall be sent over USB. I have no space to copy this data to an USB intermediate buffer.

ALABSTM, 2021-05-05:
Hello everyone,

Thank you all for your comments and feedback.

Regarding `UserTxBufferFS[]` and `UserRxBufferFS[]`, it is normal to have them in the code generated by Cube MX. Indeed, the tool is intended to generate the "skeleton" of what is intended to be a **standard** USB application.
These very buffers are the one intended to be used by the user for their application. In case the user want to define their owns, up to them to customize the generated code. From the tool perspective, however, we **shall keep** these buffers.

Regarding further uses of these two buffers, you can notice some from the [CDC_Standalone](https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32469I_EVAL/Applications/USB_Device/CDC_Standalone) application, taken as an example.

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Projects/STM32469I_EVAL/Applications/USB_Device/CDC_Standalone/Src/usbd_cdc_interface.c#L239-L241
https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Projects/STM32469I_EVAL/Applications/USB_Device/CDC_Standalone/Src/usbd_cdc_interface.c#L269

Regarding both calls below, they are necessary. Otherwise, eventual later calls to `USBD_CDC_TransmitPacket()` would fail.

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Projects/STM32469I_EVAL/Applications/USB_Device/CDC_Standalone/Src/usbd_cdc_interface.c#L124-L126

To conclude, parameter `length` of the `USBD_CDC_SetTxBuffer()` function does not define the size of the buffer. This latter is defined by the `APP_TX_DATA_SIZE` macro and **can be changed from the Cube MX GUI**.

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L857-L858

Rather, it indicates the length of the data **to be sent** when calling `USBD_CDC_TransmitPacket()`. Unfortunately, this parameter has not been described in the function's header. This will be done soon hopefully. This also explains why the `USBD_CDC_SetRxBuffer()` function does not have such one.

I hope my answers help. Otherwise, in case you have in mind some use-case you think relevant but not handled by Cube MX, please submit your proposal to the ST Community on the section dedicated to Cube MX.

With regards,

ALABSTM, 2021-05-27:
Hi,

@theadib, description of parameter `length` will be added to the header of function `USBD_CDC_SetTxBuffer()` in order to clarify its purpose and avoid any confusion.

@MichaelSweden, applications will be updated to ensure ongoing transfer has completed after each call to `USBD_CDC_TransmitPacket()`.

Thank you again for having highlighted these aspects.

With regards,

ALABSTM, 2021-05-27:
ST Internal Reference: 107728

ALABSTM, 2021-05-27:
ST Internal Reference: 107734

theadib, 2021-05-28:
Hi Ali,

*still for me the API is not clear *
*please could you join the two calls into 1(one):*
* USBD_CDC_SetTxBuffer(&USBD_Device, (uint8_t*)&UserTxBuffer[buffptr],
buffsize) and*

* USBD_CDC_TransmitPacket(&USBD_Device) *

*into e.g.*
* USBD_CDC_TransmitPacketBuffer(&USBD_Device ,
(uint8_t*)&UserTxBuffer[buffptr], buffsize)*

*similar for theReceive Calls.*

This will make the API more clearer and prevent misuse (i.e. changing
buffers while transfer is in progress)

*Regards, Adib.*
*--*

MichaelSweden, 2021-05-29:
The length and buffer setup with USBD_CDC_SetTxBuffer is used in USBD_CDC_TransmitPacket. These two function calls are very tight connected with each other. I suggest that both USBD_CDC_SetTxBuffer and USBD_CDC_TransmitPacket should get a better description in the usbd_cdc.c file. Please explain the relation between them. One is to pass in the arguments, the other for trigger the action.

The UserTxBufferFS buffer isn't needed. If you for any reason want to keep it, please set the default size to zero (or one, if zero isn't posssible).

I would recommend to set the default size of UserRxBufferFS to CDC_DATA_FS_OUT_PACKET_SIZE.

The "USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserTxBufferFS, 0);" call in CDC_Init_FS is strange. Here you setup to send zero bytes and supply a pointer to the buffer where the bytes can be found. Zero bytes! To send zero bytes will be to send nothing. So nothing will be read from the buffer, i.e. the pointer isn't needed. It is actually very confusing to pass in a pointer in this situation. I recommend to set this argument to NULL. This call is good to have because it will initialize the two corresponding local variables to NULL and zero (they aren't initialize elsewhere). Note that if you do another change (I will describe next), this call can be removed.

Now about the pair **USBD_CDC_SetTxBuffer and USBD_CDC_TransmitPacket**, with a proposal. These can be **combine into one function** (as already written in this issue). Move the arguments from the first and put into the second. Remove the first function and only keep USBD_CDC_TransmitPacket. This by **keep the functionality** as today. This will make it more clear what it do and how it works. Please note that it still should be a asynchronous function, i.e. return quickly. It shouldn't stall. If the user want to wait for the transmition to finish, the callback function is available. By doing this change it will also be clear that the UserTxBufferFS buffer is never used. However that is in the cdc example app. The change to migrate these two functions to one is in the API to the drivers, i.e. effect many (to the better).

The pointer (buffer) setup in USBD_CDC_SetRxBuffer is used in USBD_CDC_ReceivePacket. Please note that this call will prepare the OUT Endpoint for reception, i.e. it will not actually do the reception. The buffer pointed to by this pointer needs to still be available after this call. The pointer sent in need to point to a buffer with space for atleast CDC_DATA_FS_OUT_PACKET_SIZE bytes. This isn't documented in usbd_cdc.c file at the USBD_CDC_SetRxBuffer function.

The row "USBD_CDC_SetRxBuffer(&hUsbDeviceFS, &Buf[0]);" in CDC_Receive_FS function isn't needed. I guess the reason for it to be where is to indicate that it is possible to change the receive buffer, i.e. where the next data will be written to. For example if you like to implement a circular buffer. If you like to keep this row, I suggest to have it comment away, maybe like this:
/* USBD_CDC_SetRxBuffer(&hUsbDeviceFS, &Buf[newWriteIndex]);  // Set new write position, for next data package. */

Please be aware of that usbd_cdc_if.c is only an example. This compare to some stuff we talking about which is on the other side of the API. This belong to the driver part and is much more important (USBD_CDC_* functions).

The usbd_cdc_if.c file contains half done example and maybe that is why it is so confusing. Note that code in this file are within "USER CODE" comments and exepected to be changed by the developer. Or you can actually skip this file and implement your own send and receive functions. Maybe it would be good with a better and longer description in top of this file.

PS. Please note that my example I wrote earlier about and the way of checking for the transmition to finish. Could be implemented differently by using the CDC_TransmitCplt_FS callback function. I didn't use it because it isn't available in the F1 version of this software (I began with a F103 device, moved over to a F401 and will go for a G0). Excuse me for showing this. However both way is working and checking the variable like I do, is actually also done usbd_cdc_if.c. 

_Thank you STM. The framework that can be created with your tool make it easy to get started with your MCU's. This with the cheap china boards make it possible to create a prototype quickly. It have been fun doing this and now I'm designing my own board._

MichaelSweden, 2021-05-29:
> @MichaelSweden, applications will be updated to ensure ongoing transfer has completed after each call to `USBD_CDC_TransmitPacket()`.

@ALABSTM I'm little concerned about this you wrote, it isn't clear for me. Is your intention to change the implementation of the USBD_CDC_TransmitPacket function or change example code that calling this function?

closed 2022-04-19 by ALABSTM