STMicroelectronics/STM32CubeF4 #62: Incorrect error code assignment in SMBus driver

karnpinyo, 2021-02-22:
https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c#L1709

The error code should be `HAL_SMBUS_ERROR_PECERR` instead.

ASELSTM, 2021-02-23:
Hi @karnpinyo,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-02-23:
ST Internal Reference: 101953

ASELSTM, 2021-08-16:
Hi @karnpinyo,

The issue you reported has been fixed in the frame of version v1.26.2 of the STM32CubeF4 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM