STMicroelectronics/STM32CubeF4 #63: HAL_I2C_Mem_Read_IT routine causing interrupts to hang processor

jasonbraley, 2021-02-23:
**Describe the set-up**
 * Custom board with STM32F417, external 8MHz crystal, system clock running at 168MHz
 * External battery with BQ27750 chipset connected to main processor on I2C1
 * Compiler: gcc-arm-none-eabi-7-2018-q2-update

**Describe the bug**
While doing an interrupt-driven I2C memory read operation, a 2-byte read from a device resulted in 137 individual interrupts.

**How To Reproduce**
1. Indicate the global behavior of your application project.
Multi-task FreeRTOS application that runs a periodic motion control loop and U/I. One task updates the battery indicator by polling every second to read multiple values from the TI BQ27750, using HAL_I2C_Mem_Read_IT. When this battery read happens, the other tasks in the system fail to meet their timing requirements because the processor gets bogged down servicing I2C interrupts for up to 500us

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
Driver module, specifically stm32f4xx_hal_i2c.c

3. How we can reproduce the problem.
This is very reproducible on my device. Simply do a 2-byte read from an I2C device and watch the interrupt traffic.

**Additional context**
First analysis: After transmission of the memory address to be read, the buffer interrupt is being left enabled while the device is waiting for the restart to be processed, causing the TXE bit to trigger interrupts more than 40 times. Additionally, in the case of a 2-byte read, the buffer interrupt is being left enabled after the restart bit is processed. Since the driver is waiting for the BTF flag for a 2-byte read, the RX bit will trigger interrupts repeatedly which will be no-ops as they pass through the code.  

**Screenshots**
I've got a log of all the interrupt activity complete with SR2, SR1, CR1, CR2 activity. 
https://docs.google.com/spreadsheets/d/1GoMnkRJhdg2N3yBubqI4ouFaiV02xWcZ7q9KeIoLRdk/edit?usp=sharing

ASELSTM, 2021-03-02:
Hi @jasonbraley,

Thank you for this report.

A similar issue has been already reported #61.  In I2C memory mode, once memory address written, the device should resend one start bit before switching to read mode. Calling HAL_I2C_Mem_Read_IT() generates multiple start bits, resulting in the slave device answering with a NAK which explains the numerous interruptions generated.

After checking the code, it seems that if the start bit is late informed on SW side, we can pass several time in [I2C_MemoryTransmit_TXE_BTF() ](https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_i2c.c#L5246) until start bit is set on HW side.

As a work around, would you please try to update `I2C_MemoryTransmit_TXE_BTF()`  as proposed within the issue #61 and check if you still have the same behavior, if the number of interruptions generated by the I2C decreases or not.

```diff
else if (hi2c->EventCount == 2U)
  {
    if (CurrentState == HAL_I2C_STATE_BUSY_RX)
    {
      /* Generate Restart */
      hi2c->Instance->CR1 |= I2C_CR1_START;
+     hi2c->EventCount++;
    }
    else if ((hi2c->XferCount > 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX))
```

With regards,

jasonbraley, 2021-03-17:
Hi, @ASELSTM 

I finally got a chance to try the proposed patch (I had to upgrade my code base to support 1.25.0 first), and I'm still seeing the blocking interrupt behavior even when the restart bit is not being repeatedly set. I've updated the Google Sheet that I linked to in the defect description with the two new tabs: one of the behavior on 1.25.0 before the patch and one after. The behavior looks identical to me except that now the EventCount field gets set to 3. 

If you don't mind me offering my analysis of the situation, I think the root cause of the issue is not that the restart bit gets set multiple times. While the memory read address is being transmitted, the data buffer will naturally empty into the shift register, forcing the TXE bit to be high, which will trigger the buffer interrupt. Since we have no more to transmit (we just wanted to transmit the address), there's nothing else to do while we wait for the address to finish. But since the TXE bit is still high while that's happening, the system will just interrupt over and over and over again, hanging the processor.

Similarly, once we start receiving, in the case of a 2 byte read there's no work to do in the driver during the interrupts when the RXE bit is set; the driver is instead waiting for the BTF flag interrupt (one byte in the data register, one byte in the shift register) to complete the operation. The buffer interrupt is being left enabled in this case as well, causing repeated entry into the interrupt handler (which does nothing if I'm reading it right) before returning and interrupting again.

I've patched around this issue (it reduces the interrupt traffic considerably) in my own code by disabling the buffer interrupt when the restart bit is set and re-enabling it if the XferCount is !=2. 

Please feel free to ask any more questions, and I'll gather any data that you need as soon as I can.

wylerAG, 2021-05-31:
Hi @ASELSTM 
I'm having probably a related issue:
- STM32F4 version: 1.26.1
- uC: STM32F412
- FreeRTOS V10.3.1
- DMA_1 and I2C1 in use
- 4 I2C slaves at one bus, STM32 is the bus master

After a HAL_I2C_Mem_Write_DMA(), there are sometimes 2 additional bytes transmitted on the bus.
I detected that this bug takes place, when HAL_I2C_Mem_Write_DMA() is called after a HAL_IC2_Mem_Read_DMA() or HAL_I2C_Master_Receive_DMA() call.
[image]
The input parameters are correct, so this is defenitly a STM_HAL or a HW problem.

Here a screenshot of a read + write sequence:
[image]
The read frame was called from a differerent RTOS task, then the write frame.
So I assume that there is a sequence problem in the STM32 HAL.
 
We would welcome a bugfix, since this issue leads to invalid EEPROM data in our application.

Thank you for your feedback.

wylerAG, 2021-05-31:
Hi @ASELSTM

Now we handle this bug, by sending a dummy-write-frame with HAL_I2C_Mem_Write_DMA() in advance of the valid frame with HAL_I2C_Mem_Write_DMA().

[image]

This works, reduces i2c bus performance and is ugly.
So we would welcome a bugfix on ST side.

Thank you for your feedback.

ASELSTM, 2021-09-03:
Hi @jasonbraley, @wylerAG 

Thank you for these detailed reports. Is this issue also noticed when the application is not using the FreeRTOS ?

With regards,

jasonbraley, 2021-09-19:
@ASELSTM Yes, I've run the interrupt-driven I2C MemRead in a bare metal unit test, and the observed behavior is the same. I don't think it's related to OS interactions.

wylerAG, 2021-09-21:
@ASELSTM I haven't done any NON-RTOS tests so I can't give a feedback on this. At the moment, i have no setup to test this.

ASELSTM, 2022-01-11:
ST Internal Reference: 120628

wylerAG, 2022-05-19:
Hi @ASELSTM
This issue is now fixed in the ST lib! Thanks a lot, now a can remove my ugly workaround!

    STM32F4 version: 1.7.1.3
    uC: STM32F412

ALABSTM, 2023-07-26:
Fixed in commit d5af56388ff037735ac99de39abf2b46f9921aa3

closed 2023-07-26 by ALABSTM

wylerAG, 2024-04-23:
Hi @ASELSTM
Still a sequence issue in the I2C STM32 HAL part.

**Setup:**
- STM32F4 version: 1.28.0
- STM32CubeF4\Drivers\STM32F4xx_HAL_Driver version: 1.8.2 (newer commit checked as well)
- uC: STM32F412
- FreeRTOS V10.3.1
- DMA_1 and I2C1 in use
- 4 I2C slaves at one bus, STM32 is the bus master

**Problem:**
When calling HAL_I2C_Master_Receive_DMA with a known transfer size, sometimes (0.02%) the I2C Master will transfer additional 1 or 2 frames, then blockes itself -> timeout.
[image]

To compare, this is a normal transfer -> OK:
[image]

**My debugging thoughts:**
1. HAL_I2C_Master_Receive_DMA()
2. Within this function "SET_BIT(hi2c->Instance->CR1, I2C_CR1_ACK);" will be executed
3. a task swap will occure
4. SET_BIT(hi2c->Instance->CR2, I2C_CR2_DMAEN); will be executed too late

**Possible solution:**
Setting I2C_CR2_DMAEN before I2C_CR1_ACK
This works in my experimental test scenario.
I have see many of this sequences in the STM32 HAL code, so please review this and I would be very glad on a fast and reliable bugfixing from your side.

wylerAG, 2024-04-23:
Case Number: 00204026

ALABSTM, 2024-05-27:
Hi @wylerAG,

Would you mind reporting your findings in a new issue or pull-request if not already done? This would make its tracking and processing easier.

Thank you,