STMicroelectronics/STM32CubeF4 #65: STM32F4 1.26 Vector table position can not be defined by compiler define

theadib, 2021-03-04:
@see https://github.com/STMicroelectronics/STM32CubeF4/issues/30
@see https://github.com/STMicroelectronics/STM32CubeF4/pull/64

Hello ST Team please listen to your users.

I need to be able to redefine the position of the vector table based on compiler defines.
So that I do not need to modify code when I compile for target that has bootloader.
in my proposal #30 this was possible.

now in 1.26 the template code lookes like that:
https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c
`
#if defined(USER_VECT_TAB_ADDRESS)
/*!< Uncomment the following line if you need to relocate your vector Table
     in Sram else user remap will be done in Flash. */
/* #define VECT_TAB_SRAM */
#if defined(VECT_TAB_SRAM)
#define VECT_TAB_BASE_ADDRESS   SRAM_BASE       /*!< Vector Table base address field.
                                                     This value must be a multiple of 0x200. */
#define VECT_TAB_OFFSET         0x00000000U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x200. */
#else
#define VECT_TAB_BASE_ADDRESS   FLASH_BASE      /*!< Vector Table base address field.
                                                     This value must be a multiple of 0x200. */
#define VECT_TAB_OFFSET         0x00000000U     /*!< Vector Table base offset field.
                                                     This value must be a multiple of 0x200. */
#endif /* VECT_TAB_SRAM */
#endif /* USER_VECT_TAB_ADDRESS */
`
and 
`
#if defined(USER_VECT_TAB_ADDRESS)
  SCB->VTOR = VECT_TAB_BASE_ADDRESS | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
#endif /* USER_VECT_TAB_ADDRESS */
`

1) please set the vector address register regardless of USER_VECT_TAB_ADDRESS

2) please test before setting  VECT_TAB_OFFSET
`
#if !defined(VECT_TAB_OFFSET)
#define VECT_TAB_OFFSET         0x00000000U     /*!< Vector Table base offset field.
#endif
`
3) however I do not really understand the use of USER_VECT_TAB_ADDRESS
why is it not as simple as:
`
#if defined(USER_VECT_TAB_ADDRESS)
  SCB->VTOR = USER_VECT_TAB_ADDRESS);
#else
  SCB->VTOR = VECT_TAB_BASE_ADDRESS | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
#endif /* USER_VECT_TAB_ADDRESS */
`

RKOUSTM, 2021-08-17:
Hi @theadib,

I hope you are fine. Our development teams said your proposal could not be taken into consideration for the following reason:

>The modifying of VTOR is now only performed if USER_VECT_TAB_ADDRESS is activated (in the IDE or directly in the file). This is done on purpose as the memory remap is performed at boot, allows the vector table to be basically at address 0x00. So, if the user does not want to relocate it anywhere else, there is no need to access this register. More than that some products (like dual CPU ones) use the boot address from the option bytes, so having a generic way of setting VTOR is worse than letting the default configuration (option bytes set boot @ => that one is remapped @ 0x0)

Please allow me to close this issue now and thank you for your comprehension. We are looking forward to reading from you again.

With regards,

closed 2021-08-17 by RKOUSTM

diyfr, 2021-12-17:
Same problem with IAP. 
Forced to make this modification with each code regeneration, it is rather painful. Strongly that this is put in place !!

theadib, 2022-10-28:
Hello ST Team not solved at all at the 1.27.1 release.

here a pullrequest to SOLVE the issue.
https://github.com/STMicroelectronics/STM32CubeF4/pull/146

Adib.

ALABSTM, 2025-02-19:
Fixed in commit 5d5252a044bcd62fd3197cd3bcadc754b1c1b97b

ALABSTM, 2025-02-19:
ST Internal Reference: 199172