STMicroelectronics/STM32CubeF4 #67: Hardcoded address in use

CanastraRF, 2021-03-23:
In stm32f4xx_hal_flash.h is a Hardcoded address in use

```
#define __HAL_FLASH_SET_LATENCY(__LATENCY__) (*(__IO uint8_t *)ACR_BYTE0_ADDRESS = (uint8_t)(__LATENCY__))
#define ACR_BYTE0_ADDRESS           0x40023C00UL

```
I think it is better to use the following code
`#define __HAL_FLASH_SET_LATENCY(__LATENCY__) (FLASH->ACR = (uint8_t)(__LATENCY__))
`
Can You explain your motivation?

Also other addresses shoud replaced / removed.

```
#define OPTCR_BYTE0_ADDRESS         0x40023C14UL
#define OPTCR_BYTE1_ADDRESS         0x40023C15UL
#define OPTCR_BYTE2_ADDRESS         0x40023C16UL
#define OPTCR_BYTE3_ADDRESS         0x40023C17UL

#define RCC_CR_BYTE2_ADDRESS       0x40023802UL
#define RCC_CIR_BYTE1_ADDRESS      ((uint32_t)(RCC_BASE + 0x0CU + 0x01U))
#define RCC_CIR_BYTE2_ADDRESS      ((uint32_t)(RCC_BASE + 0x0CU + 0x02U))
#define RCC_BDCR_BYTE0_ADDRESS     (PERIPH_BASE + RCC_BDCR_OFFSET)

```
Thanks
ReFe

RKOUSTM, 2021-04-02:
Hi @CanastraRF,

Thank you for your contribution. Your report will be forwarded to our development team. We will get back to you as soon as we have more details.

Thank you once more for this contribution.

With regards,

RKOUSTM, 2021-09-01:
Hi @CanastraRF,

Thank you for your contribution. Regarding the first point, an internal tracker has been logged and a fix will be implemented and made available in the frame of a future release to update the definitions of the flash set latency. However, for the second point, actually your request can't be accepted in order to not break the compatibility of our HAL drivers and that Customers use the oldest version of our drivers. 

With regards,

RKOUSTM, 2021-09-01:
ST Internal Reference: 112635

RKOUSTM, 2021-09-30:
Hi @CanastraRF,

I hope you are doing well. According to our development teams, the proposed enhancement in the hal_flash.h file to align with other STM32 series is not possible to implement because the DCRST and the ICRST bits in the FLASH access control register (FLASH_ACR) are read only:

[image]

Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue. Thank you for your comprehension and thank you again for your contribution.

We are looking forward to reading from you again.

With regards,

closed 2021-09-30 by RKOUSTM

CanastraRF, 2021-09-30:
Hi @RKOUSTM 
Sorry I did not understand your explanation.
**#define __HAL_FLASH_SET_LATENCY(__LATENCY__) (*(__IO uint8_t *)ACR_BYTE0_ADDRESS = (uint8_t)(__LATENCY__))**
and 
**#define __HAL_FLASH_SET_LATENCY(__LATENCY__) (FLASH->ACR = (uint8_t)(__LATENCY__))**
results in absolutely the same code.
In both case is one byte written to ACR. This has now difference to DCRST or ICRST.
The improvement of my version is, that no additional address is required and everything is depend of the FLASH struct.
BTW.
We often emulate STM32 code in a windows app. 
In this case we replace the 
_#define FLASH               ((FLASH_TypeDef *) FLASH_R_BASE)_
with 
_#define FLASH               (&sFLASH)_
and 
_FLASH_TypeDef              sFLASH_
Now read, write access the periphery struct is possible.
But all hard coded address like ACR_BYTE0_ADDRESS must be handled separat. This should not be required.
So please replace all hard coded address with the exist struct element.