STMicroelectronics/STM32CubeF4 #68: HAL USART Driver Error for STM32F4xx

syed0mer, 2021-03-23:
We are a startup working with STM32CubeF429 Discovery Kit. We downloaded the official STM32CubeIDE and generated code for USART peripheral. 

On compiling the code, we found an error in the driver file: stm32f4xx_hal.c in the following functions at bottom of this file: a) 'USART_Receive_IT()' and b) 'USART_TransmitReceive_IT()'. 

The error is minor that defined macros used in the body of these functions in the if-else block should contain 'USART' instead of 'UART' in them. As the files are for USART. This throws the error. As soon as we corrected this, the project compiled successfully. 

I thought the ST community should know as drivers should not be modified in general. Thank you. See the individual files as well.

[STM32CubeIDE- CubeHALF429 USART Driver Files.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6192046/STM32CubeIDE-.CubeHALF429.USART.Driver.Files.zip)

ASELSTM, 2021-03-24:
Hi @syed0mer,

Thank you for this report. Indeed, the issue you pointed out is confirmed, an internal tracker has been logged and a fix will be implemented and made available as soon as possible. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-03-24:
ST Internal Reference: 103712

syed0mer, 2021-03-29:
Acknowledged. Thanks.

ASELSTM, 2021-04-28:
Hi @syed0mer,

The issue you reported has been fixed in the frame of version v1.26.1 of the STM32CubeF4 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-04-28 by ASELSTM