STMicroelectronics/STM32CubeF4 #69: PLL can't be updated when PLL is already used for sysclock

SibertEnovates, 2021-04-01:
**Setup**
Custom board containing STM32F439VI with external 8MHz crystal
FW_F4 V1.26.1
Toolchain: SW4STM32
Cube MX version: 6.2.0

**Bug**
Bug occurs when setting the PLLCLK as source for the SYSCLK in CubeMX, using 8MHz HSE as source for PLL, as shown below.
[image]

**Analysis**
The `HAL_RCC_OscConfig(&RCC_OscInitStruct)` call in main.c (in `SystemClock_Config`) fails. This failure is caused because the (I assume by default) input for SYSCLK is already set to PLLCLK, which is checked on line 467 of `stm32f4_hal_rcc.c` (`if(__HAL_RCC_GET_SYSCLK_SOURCE() != RCC_CFGR_SWS_PLL)`. This check is done because the code within that if disables the PLL, which is not possible according to the reference manual when SYSCLK = PLLCLK.
Because of this, PLL is not set to the new settings defined in `SystemClock_Config` but rather goes to either line 544 or 552  ( `stm32f4_hal_rcc.c` )where the current PLL settings are checked against the user settings which obviously fails because the user settings are impossible to write in this case.

The issue can be circumvented by prepending `SystemClock_Config` with a call that sets the SYSCLK to HSI instead of PLLCLK.
I feel like this is a bug rather than expected behavior.

Glaciuse, 2021-04-02:
To temporary fix add
__HAL_RCC_SYSCLK_CONFIG(RCC_CFGR_SWS_HSI);

before calling
HAL_RCC_OscConfig

RKOUSTM, 2021-04-02:
Hi @SibertEnovates,

First we would like to thank you for your contribution. 

We tried to reproduce the issue you described, but we couldn't notice the problem. In order to allow a better analysis of your problem, could you please give us the .ioc file that you are used to reproduce this issue,  so that we can give you an answer.  

Thank you again for your contribution.

With regards,

chrisreedio, 2021-04-04:
Using STM32CubeMX 6.2.1 with a Nucleo F030R8 running STM32Cube FW_F0 1.11.2.

I had been getting a error in the following line in SystemClock_Config on
all my Nucleo cores with a STOCK unmodified project generated from CubeMX.

  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

Following @Glaciuse's tip, I added
__HAL_RCC_SYSCLK_CONFIG(RCC_CFGR_SWS_HSI);
in the User Code Begin section above SystemClock_Config() and the hard
fault is no longer being thrown.

The interesting thing is, before I added this line, Any of my Nucleo boards
would hard fault when set to use the internal PLL driven from the HSI (HSE
not populated).
If I commented out the Error_Handler(); call so that it wouldn't trigger
the hard fault, the code would actually still work just fine (the best I
can tell).
Also, switching the clock source in CubeMX to HSI instead of PLL driven by
HSI, caused the error to also disappear without adding the line of code.

This is my first message to the list after lurking for years so hopefully I
followed proper edicate.
Hope this testing helps in some way!

*Christopher Reed*
Senior Software Developer / Embedded Systems Engineer
Nashville, TN

chrisreedio, 2021-04-04:
Using STM32CubeMX 6.2.1 with a Nucleo F030R8 running STM32Cube FW_F0 1.11.2.

I had been getting a error in the following line in SystemClock_Config on
all my Nucleo cores with a STOCK unmodified project generated from CubeMX.
```
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
```
Following @Glaciuse's tip, I added
```__HAL_RCC_SYSCLK_CONFIG(RCC_CFGR_SWS_HSI);```
in the User Code Begin section above SystemClock_Config() and the hard
fault is no longer being thrown.

The interesting thing is, before I added this line, Any of my Nucleo boards
would hard fault when set to use the internal PLL driven from the HSI (HSE
not populated).
If I commented out the Error_Handler(); call so that it wouldn't trigger
the hard fault, the code would actually still work just fine (the best I
can tell).
Also, switching the clock source in CubeMX to HSI instead of PLL driven by
HSI, caused the error to also disappear without adding the line of code.

Hope this testing helps in some way

Glaciuse, 2021-04-05:
> Hi @SibertEnovates,
>
> First we would like to thank you for your contribution.
>
> We tried to reproduce the issue you described, but we couldn't notice the problem. In order to allow a better analysis [...]

Here is minimal .ioc and schematics

Thank you!

[stmc1.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6256786/stmc1.zip)

RKOUSTM, 2021-04-07:
Hi @Glaciuse, 

Thank you for your contribution.  Concerning the .ioc received file, we generated a project using the STM32CubeMX v6.2.1 and the STM32CubeIDE v1.6.0 running on [STM32F429ZIT6](https://www.st.com/en/microcontrollers-microprocessors/stm32f429zi.html#overview&secondary=st_description_sec-nav-tab) board with a 25 MHZ HSE oscillator but we couldn't notice any problem. 

Regarding the HSE value, if you have not chosen the right values for the HSE oscillator on your custom board, there will be a problem on this side. 

Please let us know if it is the case and thank you once more for your contribution. Do not hesitate in case you have other questions.

With regards,

RKOUSTM, 2021-04-07:
Hi @chrisreedio,

Thank you for this report. However, there is no HSE oscillator on STM32F030R8-Nucleo board. So you need to activate the PLL with HSI/2 as source ([link](https://github.com/STMicroelectronics/STM32CubeF0/blob/4390ff6bfb693104cf97192f98c3dc9e3a7c296a/Projects/STM32F030R8-Nucleo/Examples/GPIO/GPIO_EXTI/Src/main.c#L90-L100))

I hope the link I provided you with was helpful. Thank you again for your reports. 

With regards,

SibertEnovates, 2021-04-07:
Here is my ioc-file. Sorry for the delay.
[evsimtest.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6273349/evsimtest.zip)

chrisreedio, 2021-04-07:
> Hi @chrisreedio,
>
> Thank you for this report. However, there is no HSE oscillator on STM32F030R8-Nucleo board. So you need to activate the PLL with HSI/2 as source [...]

Thanks for your reply. I have noticed there is no HSE populated by default. I encountered my reported errors while having the PLL and HSI selected.

[image]

This clock configuration gave me the error until I pasted in the extra line as I mentioned above.

RKOUSTM, 2021-04-08:
Hi @SibertEnovates, @chrisreedio,

Thank you for your feedback. 

First of all, thank you for your reply @SibertEnovates. In your .ioc file, I noticed that you chose `HSI` as the source for `PLL`. Indeed, if you need to set the PLLCLK as the source for the SYSCLK, using 8MHz HSE as the source for the PLL, you need to switch the clock source to HSE instead of PLL driven by HSI. 

We couldn't notice any problem when I generated your project using the `STM32CubeMX v6.2.1` and the `STM32CubeIDE v1.6.0` running on [STM32F439ZIT6](https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-f439zi.html) board with a 8 MHZ HSE oscillator. Please check and choose the right values for the HSE value according to your `custom board` in order not to have problems on this side. Thank you for once again for your contribution. 

Hello @chrisreedio,
 
Thank you for your report. We tried to reproduce the issue you described, but we couldn't notice the problem. Please find attached a generated template project for [STM32F030R8](https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-f030r8.html) Nucleo board using `STM32CubeMX v6.2.1` and `EWARM v8.50.6`.

[TEST_PLL.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6277655/TEST_PLL.zip)

I hope this generated template addresses your need. 

Thank you again for your report. Do not hesitate in case you have other questions. We are looking forward to reading from both of you again.

With regards,

SibertEnovates, 2021-04-12:
> First of all, thank you for your reply @SibertEnovates. In your .ioc file, I noticed that you chose `HSI` as the source for `PLL`. Indeed, if you need to set the PLLCLK as the source for the SYSCLK, [...]

Oops. I had a few days off last week so I sent the file in a hurry. Must have been the wrong one. Attached is the right one. Anyhow, the issue should exist as well when selecting HSI as PLLCLK source. 

To me, it seems like a bug that you manually have to change from PLLCLK to HSI or HSE as SYSCLK source to actually use the PLL. The generated HAL should take care of this, this is in my opinion expected behavior and more user friendly.

[evsimtest.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6294666/evsimtest.zip)

RKOUSTM, 2021-04-29:
Hi @SibertEnovates, 

We tried once more to reproduce the problem you have described based on a regenerated project of your sent .ioc file running on our [STM32F439ZIT6](https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-f439zi.html) board, with no success. I really hope you could find the root cause of this problem and, ideally, solve it.

Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2021-04-29 by RKOUSTM

jorisdobbelsteen, 2023-03-13:
If you are using OpenOCD, it might be that you are running into the issue where the debugger causes this. Please check stm32f4x.cfg. The issue is that the debugger configures the system to 64 MHz with the PLL & HSI:
```
$_TARGETNAME configure -event reset-init {
	# Configure PLL to boost clock to HSI x 4 (64 MHz)
	mww 0x40023804 0x08012008   ;# RCC_PLLCFGR 16 Mhz /8 (M) * 128 (N) /4(P)
	mww 0x40023C00 0x00000102   ;# FLASH_ACR = PRFTBE | 2(Latency)
	mmw 0x40023800 0x01000000 0 ;# RCC_CR |= PLLON
	sleep 10                    ;# Wait for PLL to lock
	mmw 0x40023808 0x00001000 0 ;# RCC_CFGR |= RCC_CFGR_PPRE1_DIV2
	mmw 0x40023808 0x00000002 0 ;# RCC_CFGR |= RCC_CFGR_SW_PLL

	# Boost JTAG frequency
	adapter speed 8000
}
```

Blargian, 2024-10-21:
> If you are using OpenOCD, it might be that you are running into the issue where the debugger causes this. Please check stm32f4x.cfg. The issue is that the debugger configures the system to 64 MHz [...]

@jorisdobbelsteen I seem to be having this exact problem. Is `stm32f4x.cfg.` something that can be edited such that the debugger matches CubeMX set frequencies?