STMicroelectronics/STM32CubeF4 #72: USB CDC broken under Linux

WilheJo, 2021-04-22:
5.10.30 (and all recent 5.11, 5.12) breaks CDC (or the CDC stack breaks Linux... you're free to choose).

The only error message on opening of the device:
`
[ 1333.790160] cdc_acm 1-10:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
`

So 5.10.30 does not work, 5.10.26 is fine.
This is verified with 1.25 and 1.26.1 firmware on different hardware and different firmware.

Since these releases are out for some time, I guess there's something to fix in the STM32 stack and add some quirks to be added to the linux drivers to make all that firmware out in the wild work again...

I already posted a bugreport on bugzilla: https://bugzilla.kernel.org/show_bug.cgi?id=212751

lsusb dump under 5.10.30:
```
Bus 001 Device 016: ID 0483:5740 STMicroelectronics Virtual COM Port
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         2 Abstract (modem)
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x5740 Virtual COM Port
  bcdDevice            2.00
  iManufacturer           1 STMicroelectronics
  iProduct                2 STM32 Virtual ComPort
  iSerial                 3 204E37773453
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered```

WilheJo, 2021-04-23:
fix is already in usb-next.

If you're affected, the related patch is provided in the bugzilla entry:
https://bugzilla.kernel.org/show_bug.cgi?id=212751

closed 2021-04-23 by WilheJo

ALABSTM, 2021-04-26:
Hello @WilheJo,

Thank you for this report. The link you provided redirects to a bug description indicating SHA-1 of two commits. Is this related to our source files or yours? In case it is related to ours, would you please provide more precise description about the bug and the fix? Thank you again.

With regards,

WilheJo, 2021-04-26:
It seems that there are several recent Linux Kernel releases which have broken ACM drivers.

This report has been filed since it was unclear if the USB CDC implementation on the STM32 or the Linux Kernel trigger/contain a bug.

The commits referenced in the linked bug report above contain some description on the origin of the bug. It seems that some refactoring due to circular dependencies have led to this problem.

In the end, there's a good likelihood that some of the several releases (5.10.28-5.10.32 are definitly effected; the bug is/was also present in the 5.11 and 5.12 kernel tree) will cause troubles to users. Hence I linked the bugzilla report which references the patch which can be cherrypicked from the usb-next tree to recompile a fixed kernel.

Regards

ALABSTM, 2021-04-28:
Hi @WilheJo,

Thank you for all these details and for having reported. If I understand correctly, this issue is rather related to the Linux Kernel rather than to our USB CDC implementation. As a fix, users can cherry-pick the patch from the [usb-next](https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/usb.git/log/?h=usb-next) branch.

Thank you again for this initiative. Users facing the problem you reported could land here and find the fix you proposed.

With regards,