STMicroelectronics/STM32CubeF4 #73: MMC_InitCard

dderiso, 2021-05-03:
Thank you for the time and work your team puts into these libraries. Below is a possible bug and fix that warrants further investigation. 

This section results in a command response timeout:`SDMMC_ERROR_CMD_RSP_TIMEOUT`.

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_mmc.c#L2678-L2689

As described in this [forum post](https://community.st.com/s/question/0D53W00000cSjbKSAS/fw-v1161-broke-hal-mmc-driver), reversing the order of calls resolves the issue.

  ```
/* Select the Card */
  errorstate = SDMMC_CmdSelDesel(hmmc->Instance, (uint32_t)(((uint32_t)hmmc->MmcCard.RelCardAdd) << 16U));
  if(errorstate != HAL_MMC_ERROR_NONE)
  {
    return errorstate;
  }

/* Get CSD parameters */
  if (HAL_MMC_GetCardCSD(hmmc, &CSD) != HAL_OK)
  {
    return hmmc->ErrorCode;
  }
```

ALABSTM, 2021-05-24:
Hi @dderiso,

Thank you for the report, the fix proposal, and the reference to the post on the community. Your request has been forwarded to our development teams. We will get back to you as soon as we receive their feedback.

With regards,

ALABSTM, 2021-05-27:
Hi @dderiso,

Our development teams confirmed your point. A fix will be published in a future release. Thank you again for having reported.

With regards,

RKOUSTM, 2022-03-11:
Hi @dderiso,

Thank you for your contribution. This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

Thank you again for your contribution.

With regards

closed 2022-03-11 by RKOUSTM