STMicroelectronics/STM32CubeF4 #74: RTC initialization (defect)

AntonNeutron, 2021-05-08:
STM CubeIDE 1.6.1
STM32F411
FW_F4 V1.26.1
Driver LL

No string added when initializing date "RTC_DateStruct.Day = xx;"

=================Generation code=============================
if(LL_RTC_BAK_GetRegister(RTC, LL_RTC_BKP_DR0) != 0x32F2){
  RTC_TimeStruct.Hours = 1;
  RTC_TimeStruct.Minutes = 1;
  RTC_TimeStruct.Seconds = 2;
  LL_RTC_TIME_Init(RTC, LL_RTC_FORMAT_BCD, &RTC_TimeStruct);
  RTC_DateStruct.WeekDay = LL_RTC_WEEKDAY_MONDAY;
  RTC_DateStruct.Month = LL_RTC_MONTH_JANUARY;
  RTC_DateStruct.Year = 1;
  LL_RTC_DATE_Init(RTC, LL_RTC_FORMAT_BCD, &RTC_DateStruct);
    LL_RTC_BAK_SetRegister(RTC,LL_RTC_BKP_DR0,0x32F2);
  }
==========================================================

ALABSTM, 2021-05-24:
Hi @AntonNeutron,

In your comment you mention `RTC_DateStruct.Day`. However, in you code snippet, there is `RTC_DateStruct.WeekDay`. Moreover, you speak about the addition of a string of characters, if I understood well. Would you mind refining more your description?

Thanks,

AntonNeutron, 2021-05-31:
Hi @ALABSTM,
 `RTC_DateStruct.WeekDay` just indicates the day of the week. 
If you skip the RTC_DateStruct.Day line in the initialization, then the RTC counting register is not initialized with the date of the month and will count from day zero. Thanks

ALABSTM, 2021-06-08:
Hi @AntonNeutron,

Thank you for the clarification. May also ask whether you started from one of our examples? In such case, would you mind indicating which one? You may also have used Cube MX to generate  your code. In such case, would you mind providing us with you IOC file? Thank you in advance.

With regards,

AntonNeutron, 2021-06-11:
[test.zip](https://github.com/STMicroelectronics/STM32CubeF4/files/6637978/test.zip)

ALABSTM, 2021-08-05:
Hi @AntonNeutron,

I confirm the issue you reported. Actually, it is a **Cube MX**-related issue. Indeed, even though the date is explicitly specified in the GUI (screenshot below), the corresponding structure member `Date` is not initialized by the generated code.

[image]

The generated code is lacking the line highlighted in green below:
```diff
  RTC_DateStruct.WeekDay = LL_RTC_WEEKDAY_FRIDAY;
  RTC_DateStruct.Month = LL_RTC_MONTH_JUNE;
+  RTC_DateStruct.Day = 11;
  RTC_DateStruct.Year = 21;
  LL_RTC_DATE_Init(RTC, LL_RTC_FORMAT_BCD, &RTC_DateStruct);
```

Now, since this is related to the **ecosystem** rather than to the STM32Cube firmware, the correct place to submit your report is the [stm32cubemx](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx)-dedicated page of the **ST Community**. To make things easier, you can reference this issue in your post by including its link.

Please allow me to close this issue, then. Thank you for your comprehension and thank you again for this report.

With regards,

closed 2021-08-05 by ALABSTM