STMicroelectronics/STM32CubeF4 #75: Assertion: 'REENT malloc succeeded' failed.

Malangcow, 2021-05-20:
**Describe the set-up**

The board (either ST RPN reference or your custom board).
STM32F429I-DISCO

IDE or at least the compiler and its version.
STM32CubeIDE 1.6.1
CubeMX - Software Pack - X-CUBE-TOUCHGFX.4.16.1

**Describe the bug**
Assertion fails when rand() is called.

Following is the assertion message printed in the console.

```
: Assertion: 'REENT malloc succeeded' failed. /build/gnu-tools-for-stm32_9-2020-q2-update.20201001-1621/src/newlib/newlib/libc/stdlib/rand.c:78
```

**How To Reproduce**

1. Generate project using TouchGFX Designer.

2. UI Template used from the designer was TouchGFX Demo 3(Since the display resolution of the board is 320 by 240)

3. From the main screen add a widget(button with label)

4. Configure interaction of the button. Trigger: 'Button is clicked', Action: call virtual function 'button1Clicked'

5. STM32CubeIDE: Import generated project.

6. Run CubeMX from the CubeIDE. Configure peripherals, onboard LED.

7. Implement function 'button1Clicked'
   ```cpp
   void MainView::button1Clicked()
   {
     HAL_GPIO_TogglePin(LED_LD3_GPIO_Port, LED_LD3_Pin);
     HAL_GPIO_TogglePin(LED_LD4_GPIO_Port, LED_LD4_Pin);
     
     // Set Random Color
     auto randColor = rand();
     dlogln("%s: rand = %08Xh", __PRETTY_FUNCTION__, randColor);
     uint8_t rgb_r = randColor & 0xFF;
     uint8_t rgb_g = (randColor >> 8) & 0xFF;
     uint8_t rgb_b = (randColor >> 16) & 0xFF;
     buttonWithLabel1.setLabelColor(touchgfx::Color::getColorFrom24BitRGB(rgb_r, rgb_g, rgb_b));
     buttonWithLabel1.invalidate();
   }
   ```
   
8. Build project. Program firmware binary.

9. Run the board.

10. Click the button widget.

11. Assertion fail message is printed and TouchGFX task halts.

**Screenshots**

[image]

ASELSTM, 2021-06-10:
Hi @Malangcow,

Thank you for your contribution and for this detailed description.

We have followed the steps you described and unfortunately we were not able to reproduce the issue. Could you please share the whole project if possible.

With regards,

Quphoria, 2021-08-09:
I have had this issue occur as well.
Seems to be something to do with this:
`http://sourceware-org.1504.n7.nabble.com/REENT-CHECK-VERIFY-calls-assert-func-even-if-NDEBUG-is-defined-td628431.html`

It occurred for me on:
- GNU Tools for STM32 (9-2020-q2-update)
- GNU ARM Embedded (9-2020-q2-update)

It didn't occur for me on:
- GNU ARM Embedded (7-2018-q2-update)

ASELSTM, 2021-09-03:
Hi @Quphoria,

The shared link is unfortunately not accessible.

With regards,

savke123, 2021-09-24:
HI @Malangcow try to add this line of code "srand(time(NULL));" before calling rand() function, it worked for me.

ASELSTM, 2022-09-30:
Hi @Malangcow,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2022-09-30 by ASELSTM