STMicroelectronics/STM32CubeF4 #77: SPI_WaitFlagStateUntilTimeout hangs in DMA Interrrupt

thehilde, 2021-06-04:
Set-up:
STM32F415 True Studion 9.3

I receive a SPI signal with DMA in SPI_DIRECTION_2LINES_RXONLY Mode

SPI_WaitFlagStateUntilTimeout(hspi, SPI_FLAG_RXNE, RESET, Timeout, Tickstart) != HAL_OK) 
hang...

The SPI_FLAG_RXNE don't release so the timeout sould fix the problem. But  timeout  don't work because he is in the DMA interrrupt and HAL_GetTick() didn't count up.

ASELSTM, 2021-06-15:
Hi @thehilde,

Thank you for this report. If the issue is systematically reproducible on your side, would you please give us more details on how to reproduce it and if possible share the project so that we can reproduce the same behavior and give you an answer.

With regards,

ASELSTM, 2021-08-23:
Hi @thehilde,

Please allow me to close this thread as no activity has been registered since a couple of month. You can reopen it at any time if you have details to provide us to help you solve this issue. Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-08-23 by ASELSTM