STMicroelectronics/STM32CubeF4 #78: uwTickPrio not set when using alternate system timebase source

akudlacek, 2021-06-08:
I tested on STM32F429I-DISC1 board and one of my own design.
IDE: STM32CubeIDE V1.6.1
MCU: STM32F429ZITx
FW Package: STM32Cube FW_F4 V1.26.1

- Assert fails when checking the NVIC preemption priority of system timebase source.
- Point of failure detected at: main() / SystemClock_Config() / HAL_RCC_ClockConfig() / HAL_InitTick() / HAL_NVIC_SetPriority() / assert_param(IS_NVIC_PREEMPTION_PRIORITY(PreemptPriority));
- uwTickPrio is initally set to an invalid priority and is never changed to a valid one.

**How To Reproduce**
1. Default cubemx project with all peripherals in their default mode.
2. Full assert must be enabled and I put asm("BKPT"); within the assert_failed function in main.c
3. Debug run and it will crash

**Additional context**
System timebase source has to be different than systick.
This is not the first occurrence of this bug see:
[STM32CubeF4 closed issue](https://github.com/STMicroelectronics/STM32CubeF4/issues/18)

This has also been seen on other cubemx MCU families
[STM32CubeF1](http://ivanbayan.com/index.php/2019/09/04/stm32cube-fw_f1-v1-8-0-package-breaks-hal-time-source-init/)
[STM32CubeF7](https://github.com/STMicroelectronics/STM32CubeF7/issues/23)

A work around is to set uwTickPrio to 15 between HAL_Init() and SystemClock_Config() in main.c

I think this should be enough and since it appears to have been fixed in STM32CubeF7 I think the same could be applied here although I did not look to see what that looked like.

RKOUSTM, 2021-07-01:
Hi @akudlacek,

Thank you for your contribution. After investigation, the point you reported is related to the `CubeMX` generation problem.

This point has ben reported to `CubeMX development teams` who will take in charge the issue. 

An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release of `CubeMX`.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-07-02:
ST Internal Reference: 110035