STMicroelectronics/STM32CubeF4 #80: Touchscreen presses not detected on versions D/E of STM32F429-Discovery board

bmcdonnell-fb, 2021-07-06:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board): _is one of these the number you're referring to?_ STM32F429I-DISC1 / MB1075-F429I-E01
 * Atollic TrueSTUDIO for STM32 v9.1.0 (Build-id: 20181011-1241)

**Describe the bug**
Touchscreen does not respond to any touches.

**How To Reproduce**

Prerequisite (I think): a revision D or E board of [STM32F429I-DISC1](https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-discovery-kits/32f429idiscovery.html).

***Optional: First build and run the older version yourself, to see that it works on this newer board***

1. Obtain [STSW-STM32138](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-standard-peripheral-library-expansion/stsw-stm32138.html) (STM32F429 discovery firmware package (UM1662) v1.0.3). Unpack the zip.
2. Open the project from `Projects\Demonstration\TrueSTUDIO\STM32F429I-Discovery_Demo\` in the Atollic TrueSTUDIO IDE. (Similarly to step 3 in the next subsection - `File` &rarr; `Open Projects from File System...`, etc.)
3. In Project Properties, `C/C++ Build` &rarr; `Settings` &rarr; `Tool Settings` tab &rarr; `C Compiler` &rarr; `Symbols`, change `USE_STM32F429I_DISCO_REVC` to `USE_STM32F429I_DISCO_REVD`. (Per `stm32f429i_discovery.h`, `USE_STM32F429I_DISCO_REVD : for RevD or RevE boards`.)
4. Build, load, run. (Find it at `Projects\Demonstration\TrueSTUDIO\STM32F429I-Discovery_Demo\Debug\STM32F429I-Discovery_Demo.elf`.) It appears to be the same firmware that was loaded out-of-the-box. Touchscreen works as expected.

***Build demo firmware from this repo to produce the problem***

1. Clone this repo
2. As hinted by [this readme](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Middlewares/ST/STemWin/readme.txt), obtain [STemWIN-MW](https://www.st.com/en/licensed-software/stemwin-mw.html) version 5.44.40. (Note, the later 5.44.70 does not include the library files for STM32F4.) Extract the STemWin zip, and copy the contents to the `Middlewares/ST/STemWin/` subdir in your working copy of this repo.
3. Import the project as follows. In Atollic TrueSTUDIO IDE, do `File` &rarr; `Open Projects from File System...`. Click `Directory...`; navigate to the `Projects\STM32F429I-Discovery\Demonstrations\STemWin\SW4STM32\STM32F429I-Discovery\` subdir of your working copy of this repo. Click `Finish`, and `OK` the conversion.
[image]
[image]
4. Build the `STM32F429I-Discovery` project.
5. Use e.g. STM32CubeProgrammer to flash the resulting `STM32F429I-Discovery.elf` onto the micro. (Find it in the working copy under `Projects/STM32F429I-Discovery/Demonstrations/STemWin/SW4STM32/STM32F429I-Discovery/Debug/`.)
6. If prompted, use a pointy object to calibrate the touchscreen.
7. Observe the application at the homescreen. Try and get the touchscreen to respond to any presses; it doesn't.

***Suspected problem***

As pictured below, the BSP in the [old demo firmware](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-standard-peripheral-library-expansion/stsw-stm32138.html) has been updated to account for later board revisions (D/E). The firmware in this repo has not.

[image]

**Additional context**
I believe this STM32CubeF4 repo utilizes an old version of the [STM32 BSP](https://github.com/STMicroelectronics/stm32-bsp-common), namely [v4.0.1](https://github.com/STMicroelectronics/stm32-bsp-common/releases/tag/v4.0.1).

The [old demo firmware (STSW-STM32138)](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-standard-peripheral-library-expansion/stsw-stm32138.html) appears to be based on something even older.

This repo has many demo and applications; the old demo firmware has less, and is based on older code. An update here would be useful.

**Screenshots**
(None to add)

bmcdonnell-fb, 2021-07-07:
When I apply the board-rev-specific changes from the old demo firmware to this repo, it responds as expected for a minute or so, then the app seems to hang.

[image]

ASELSTM, 2023-04-06:
Hi @bmcdonnell-fb,

The issue has been fixed in the frame of this [commit](https://github.com/STMicroelectronics/STM32CubeF4/commit/d7fef245050064d9a259b73b79ba593da9f3cf43). Please allow me this to close this thread.

With regards,