STMicroelectronics/STM32CubeF4 #82: LL_ADC_IsActiveFlag_MST_EOCS does not compile

adroudrou, 2021-07-28:
LL_ADC_IsActiveFlag_MST_EOCS does not compile on my compiler because the parameter is unused.

I think there maybe is a problem on this line
  return (READ_BIT(ADC1->SR, LL_ADC_FLAG_EOCS) == (LL_ADC_FLAG_EOCS));

maybe use ADCxy_COMMON-> insted of ADC1->

ASELSTM, 2021-07-28:
ST Internal Reference: 111256

M-Bab, 2021-08-04:
I am quite sure this is wrong. Maybe this is the right solution?

```diff
diff --git a/soup/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h b/soup/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h
index 93489114..8df560d2 100644
--- a/soup/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h
+++ b/soup/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h
@@ -4395,7 +4395,7 @@ __STATIC_INLINE void LL_ADC_ClearFlag_AWD1(ADC_TypeDef *ADCx)
   */
 __STATIC_INLINE uint32_t LL_ADC_IsActiveFlag_MST_EOCS(ADC_Common_TypeDef *ADCxy_COMMON)
 {
-  return (READ_BIT(ADC1->SR, LL_ADC_FLAG_EOCS) == (LL_ADC_FLAG_EOCS));
+  return (READ_BIT(ADCxy_COMMON->CSR, LL_ADC_FLAG_EOCS_MST) == (LL_ADC_FLAG_EOCS_MST));
 }
```

ASELSTM, 2022-03-11:
Hi @adroudrou,

This issue has been fixed in the frame of version v1.27.0 of the STM32CubeF4 published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM