STMicroelectronics/STM32CubeF4 #83: stm32f4xx_hal_adc.c ADC startup code contains a delay loop that's x10 slower than it should be

InconsolableCellist, 2021-07-28:
I'm using an STM32F407VGT6 on an STM32F4DISCOVERY board, running at 168 MHz, with an ADC in IT mode using the HAL libraries (STM32CubeM4 `master` 2f3b26f16559f7).

Calling `HAL_ADC_Start_IT(&hadc1)` takes ~30 uS. The STM32F407 datasheet ("DocID022152 Rev 8"), p. 134 defines the t_STAB power-up time for the ADC as 2-3 uS.

There's a loop at `stm32f4xx_hal_adc.c` line 1061-1065 that's intended to delay the CPU by `ADC_STAB_DELAY_US` microseconds (`#define`d to `3`uS for my configuration, stm32f4xx_hal_adc.h:640):

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc.c#L1060

In reality, even with optimizations like `-Ofast`, this loop takes ~30 uS in practice, and must necessarily do so, because `LDR Reg, [mem]` alone takes at least two cycles on an M4.

Replacing the counter value with: 

`    counter = (ADC_STAB_DELAY_US * (SystemCoreClock / 1000000U / 10)); `

produces a delay of about 3.625 uS for my environment, which is much closer to the spec. 

This problem is repeated on lines 740, 1385, and 1760, and the same design pattern appears in `stm32f4xx_hal_adc_ex.c` lines 168, 271, and 622:

https://github.com/STMicroelectronics/STM32CubeF4/blob/2f3b26f16559f7af495727a98253067a31182cfc/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc_ex.c#L168

Ideally all of these should be replaced with something like a macro that implements a uS delay function, possibly by unrolling a volatile ASM block of known cycle-count and looping over it the proper number of times, with a #pragma to prevent optimization.

ASELSTM, 2021-09-16:
Hi @InconsolableCellist,

We do actually not support the us delays in our Firmware package, only ms delays are supported. However, we will consider your request.  We will  study this solution and check its feasibility. Though, this may take time so thank you for your contribution and for your comprehension. 

With regards,

ASELSTM, 2021-09-16:
ST Internal Reference: 113314

ALABSTM, 2024-01-30:
Hi @InconsolableCellist,

I hope you are fine. Back to this point you reported. There is currently no plan to add a macro or a function implementing delays in microseconds. I will let you know should there be any update.

In the meanwhile, the recommended solution is to use a counter variable decremented periodically as done in the code [snippet](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/89956295d29f6b79885c0f78847558141d79d89a/Src/stm32f4xx_hal_adc.c#L737-L742) below:

```c
    /* Compute number of CPU cycles to wait for */
    counter = (ADC_STAB_DELAY_US * (SystemCoreClock / 1000000U));
    while (counter != 0U)
    {
      counter--;
    }
```

Please allow me to close this issue. Thank you for your comprehension and thank you again for your proposal.

With regards,

closed 2024-01-30 by ALABSTM