STMicroelectronics/STM32CubeF4 #86: HAL_SMBUS_Master_Receive_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC) does not always drive SCL

RhinoBabies, 2021-08-18:
**Describe the set-up**
Custom board with STM32F439ZI
STM32CubeIDE V1.7.0
HAL Driver Version V1.7.13

**Describe the bug**
SCL is not being driven by a call to HAL_SMBUS_Master_Receive_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC) when it follows a completed transfer from HAL_SMBUS_Master_Transmit_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC).

**How To Reproduce**
1. Indicate the global behavior of your application project.
Communicating with an INA238 chip where STM32F4 is configured as SMBus master and PEC is disabled

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_smbus.c

3. The use case that generates the problem.
HAL_SMBUS_Master_Transmit_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC)
wait for completion and handle HAL_SMBUS_MasterTxCpltCallback()
after interrupt handled, then call HAL_SMBUS_Master_Receive_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC)

**Additional context**
After the transmit is complete, the SMBUS_HandleTypeDef's PreviousState variable is being left to SMBUS_STATE_READY, but HAL_SMBUS_Master_Receive_IT(...) checks for SMBUS_STATE_NONE.

A call to the receive function works correctly and drives the SCL line when:

1. it is the initial call using the SMBus,
2. it is preceded by a transmit + receive transaction, such as 
HAL_SMBUS_Master_Transmit_IT(..., SMBUS_FIRST_FRAME)
wait for completion and handle HAL_SMBUS_MasterTxCpltCallback()
after interrupt handled, then call HAL_SMBUS_Master_Receive_IT(..., SMBUS_LAST_FRAME_NO_PEC), or 
3. when the PreviousState is set to SMBUS_STATE_NONE, e.g.
HAL_SMBUS_Master_Transmit_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC)
hsmbus->PreviousState = SMBUS_STATE_NONE;
HAL_SMBUS_Master_Receive_IT(..., SMBUS_FIRST_AND_LAST_FRAME_NO_PEC).

It looks like there could be other discrepancies related to state usage (i.e. private defines for PreviousState versus HAL_SMBUS_StateTypeDef).

Thank you,
jv

ASELSTM, 2022-01-24:
ST Internal Reference: 121446

RJMSTM, 2023-04-14:
Hello @RhinoBabies 

Thank you again for having reported this point. It has been fixed in the frame of version 1.28.0 of this firmware.

With regards,